immortalwrt/immortalwrt #1: 修改版的ssr-plus配合AdGuardHome是否有问题？

shenzt68, 2019-08-30:
[image]
[image]
只要是用GFW列表模式，翻不出，只能绕过中国ip模式才能出去！
关闭AdGuardHome，没问题

1715173329, 2019-08-30:
请具体说明翻不出是什么意思

shenzt68, 2019-08-30:
就是𣎴能浏览被屏蔽网站

1715173329, 2019-08-30:
@shenzt68 怎么不能浏览？

1715173329, 2019-08-30:
提供具体信息
是否是因为DNS解析，或者单纯的连接超时等问题
确保AdGuardHome正常解析域名

shenzt68, 2019-08-30:
连接超时

1715173329, 2019-08-30:
@shenzt68 看下是不是DNS污染 GFWList里面有没有相关域名

shenzt68, 2019-08-31:
不可能没有youtube的域名吧

1715173329, 2019-08-31:
@shenzt68 你改下DNS试试
例如ssr+使用pdnsd

zxlhhyccc, 2019-08-31:
> @shenzt68 看下是不是DNS污染 GFWList里面有没有相关域名

先绕过中国ip模式再GFW列表模式可以

shenzt68, 2019-08-31:
我这儿，这样也不行

1715173329, 2019-09-07:
Close for Infomation is not enough.
You may reopen if provide more info.

closed 2019-09-07 by 1715173329

iceheavenboy, 2021-02-03:
我用的x86 openwrt在使用AdGuardHome也有类似的问题。调试好AdGuardHome后，必须使用绕过中国大陆模式。
但是每次都是10几个小时后，不论holleworld，还是SSR plus+，还是passwall插件，就统一都会报错。
例如holleworld和ssr plus+插件，会现实尚无任何配置。passwall插件也无法运行。

ksong008, 2021-02-03:
@iceheavenboy 建议将adgdns重定向设置为重定向53端口到AdGuardHome,并设置adh上游dns 为127.0.0.1:53