immortalwrt/immortalwrt #8: 編譯MINIDLNA SFTP-SERVER此兩套件會報錯

kai456789, 2019-09-27:
點選MINIDLNA
FFMPEG 會出現openssl is incompatible with the gpl and --enable-nonfree is not specified.問題

點選 SFTP-SERVER會出現OPENSSL-SSH錯誤
[image]

1715173329, 2019-09-27:
@kai456789 请更新feeds源

1715173329, 2019-09-27:
有关openssh问题，我这边未出现问题；请检查您系统的GCC/G++/CLANG版本号。

kai456789, 2019-09-28:
更新feeds源 openssl問題已解決
openssh問題依舊 編譯 高通IQ4019 路由ACRH17  GCC版本如下 是否太舊 要用到5.0以上？
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.4' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.4)

1715173329, 2019-09-28:
如无必要，建议取消openssh，因为openwrt已经默认勾选了其他ssh组件。
openssh问题有待进一步处理，我会在国庆假期期间进行检查。

kai456789, 2019-09-28:
勾選SSH裡SFTP-SERVER 他就會編譯openssh 報錯 先取消這項 等待進一步處理 謝謝

1715173329, 2019-09-30:
openssh组件已更新，请尝试重新编译。

closed 2019-09-30 by 1715173329