immortalwrt/immortalwrt #13: 使用编译的K3固件不能在WEB页面更新固件是否是个例？

SouHong, 2019-11-12:
使用编译的K3固件不能在WEB页面更新固件，上传固件，更新后还是更新之前的版本，不知道是个例还是其它？

1715173329, 2019-11-13:
@SouHong 目前我的源码确实无法通过Web升级
蜜汁问题，找不到原因🌚

SouHong, 2019-11-13:
@1715173329 你好，我使用LEAN的源码是可以正常通过WEB升级，你的源码也同步了他的怎么会这样？

1715173329, 2019-11-13:
@SouHong 我对部分核心组件略有改动

SouHong, 2019-11-13:
@1715173329 我比较https://github.com/coolsnowwolf/lede/blob/master/target/linux/bcm53xx/base-files/lib/upgrade/platform.sh与https://github.com/project-openwrt/openwrt/blob/master/target/linux/bcm53xx/base-files/lib/upgrade/platform.sh  
发现你的源码中多269；270两行，是特意加的吗？

SouHong, 2019-11-13:
@1715173329 
我比较
https://github.com/coolsnowwolf/lede/blob/master/target/linux/bcm53xx/base-files/lib/upgrade/platform.sh

与

https://github.com/project-openwrt/openwrt/blob/master/target/linux/bcm53xx/base-files/lib/upgrade/platform.sh

发现你的源码中多269；270两行，是特意加的吗？

1715173329, 2019-11-13:
@SouHong 目前在上课🌚
这周六我看看 尽量解决此问题
感谢提醒

closed 2019-11-13 by SouHong

SouHong, 2019-11-15:
@1715173329
luci-app-k3screenctrl编译时遇到po2lmo找不到，我加入
luci-app-k3screenctrl--tools--po2lmo能正常编译。

1715173329, 2019-11-15:
```bash
sudo bash -c "curl -L 'https://github.com/project-openwrt/squashfs-tool/blob/master/po2lmo?raw=true' -o '/usr/bin/po2lmo' && chmod 0755 '/usr/bin/po2lmo'"
```

1715173329, 2019-11-15:
@SouHong 执行上面的命令，重新编译即可