immortalwrt/immortalwrt #49: 关于gotop

fuqiang03, 2020-03-06:
关于gotop
[image]
这个仅在master下修改编译成功

1715173329, 2020-03-06:
改这个有什么必要么。。

fuqiang03, 2020-03-06:
原来的会说no matching versions for quer "latest"

1715173329, 2020-03-06:
并未检索到你所说的 `no matching versions for quer "latest"` 错误，请提交相关截图。

1715173329, 2020-03-06:
> 但是会有适配的控制页面吗

本插件为命令行工具，没有LuCI

fuqiang03, 2020-03-06:
[image]

fuqiang03, 2020-03-06:
可惜了挺好看的

1715173329, 2020-03-06:
[image]
我觉得这是你自己的问题🙃

fuqiang03, 2020-03-06:
嗯嗯

fuqiang03, 2020-03-06:
arm构架?

closed 2020-03-06 by fuqiang03

1715173329, 2020-03-06:
[image]
如果你觉得有异议，那么这次我使用了OP官方最新源码为x86_64架构进行构建，没有任何问题

fuqiang03, 2020-03-06:
嗯嗯

1715173329, 2020-03-06:
关于这个问题，你可以参考下
https://github.com/golang/go/issues/27215

fuqiang03, 2020-03-06:
好的谢谢