immortalwrt/immortalwrt #131: 咨询luci问题

fuqiang03, 2020-04-12:
由于luci更新
官方使用了ACLs
有什么办法兼容以前的配置读取吗

zxlhhyccc, 2020-04-12:
> 由于luci更新
> 官方使用了ACLs
> 有什么办法兼容以前的配置读取吗

暂时等吧

1715173329, 2020-04-12:
You may simply revert the commit openwrt/luci@c099344 to make everything works as usual.
Besides, I've added the ACL files, hoping them will work fine.

closed 2020-04-13 by 1715173329

fuqiang03, 2020-04-24:
@1715173329 
遇到过这个问题吗
[image]
不可描述软件plus  和dnsmasq.d建立的文件冲突

1715173329, 2020-04-24:
手动创建下这个文件夹即可

fuqiang03, 2020-04-24:
这个文件夹是存在的

fuqiang03, 2020-04-24:
[image]
这有这样才能正常 否则崩溃 不之原因
[image]

fuqiang03, 2020-04-24:
@1715173329 
大神看看你那有这样的情况吗
全部使用主库源码

1715173329, 2020-04-24:
我萌新
没遇到过
应该是 `/etc/dnsmasq.d` 文件夹缺失

fuqiang03, 2020-04-24:
这两个文件夹都是存在的 好像是不可描述软件创建的

fuqiang03, 2020-04-24:
不知道注释掉会有什么影响
[image]