immortalwrt/immortalwrt #136: kernel 5.4 testing status

AmadeusGhost, 2020-04-16:
* [x] x86
* [x] ipq40xx
* [ ] ipq806x: init failed in some devices
* [ ] bcm27xx: need tested
* [ ] ramips: need rewrite dts and more tested
* [ ] bcm53xx: bcm has no human rights, no need to test

ElonH, 2020-04-17:
x86 kernel packages error:
* [ ] [ERROR: package/kernel/linux failed to build](https://github.com/project-openwrt/openwrt-isco/issues/643)
* [ ] [ERROR: package/kernel/mac80211 failed to build](https://github.com/project-openwrt/openwrt-isco/issues/640)
* [ ] [ERROR: package/kernel/ath10k-ct/regular failed to build](https://github.com/project-openwrt/openwrt-isco/issues/641)
* [ ] [ERROR: package/kernel/ath10k-ct/smallbuffers failed to build](https://github.com/project-openwrt/openwrt-isco/issues/642)
* [ ] [ERROR: package/feeds/packages/xtables-addons failed to build](https://github.com/project-openwrt/openwrt-isco/issues/644)
* [ ] [ERROR: package/ntlf9t/rtl8192du failed to build](https://github.com/project-openwrt/openwrt-isco/issues/645)
* [ ] [ERROR: package/kernel/mt76 failed to build](https://github.com/project-openwrt/openwrt-isco/issues/646)
* [ ] [ERROR: package/kernel/rtl8812au-ct failed to build](https://github.com/project-openwrt/openwrt-isco/issues/647)
* [ ] [ERROR: package/devel/perf failed to build](https://github.com/project-openwrt/openwrt-isco/issues/648)
* [ ] [ERROR: package/feeds/telephony/dahdi-linux failed to build](https://github.com/project-openwrt/openwrt-isco/issues/649)

ElonH, 2020-04-17:
ipq40xx kernel packages error:
* [ ] [package/devel/perf](https://github.com/project-openwrt/openwrt-kisco/issues/13)

ElonH, 2020-04-18:
ipq40xx kernel packages error:
* [ ] [package/lean/mt-drivers/mt_wifi](https://github.com/project-openwrt/openwrt-kisco/issues/15)

AmadeusGhost, 2020-04-18:
> ipq40xx kernel packages error:
> 
> * [ ]  [package/lean/mt-drivers/mt_wifi](https://github.com/project-openwrt/openwrt-kisco/issues/15)

mtwfi can only be built on kernel 4.14 and below

closed 2020-04-20 by AmadeusGhost