immortalwrt/immortalwrt #146: ERROR: package/feeds/base/mentohust failed to build

ElonH, 2020-04-23:
Package `package/feeds/base/mentohust` compiled error.

This issue is created from action.

- repo: [ElonH/opde (master)](https://github.com/ElonH/opde)
- commit: https://github.com/ElonH/opde/commit/351f306290f7c81b8ec5d9335e7157faf003e442
---

<details><summary>Log detail</summary>

 -  compile.txt

```
make[4]: Entering directory '/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1/src'
Makefile:46: warning: overriding recipe for target 'clean'
/home/runner/work/opde/opde/ctcgfw_sdk/include/package.mk:344: warning: ignoring old recipe for target 'clean'
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c mentohust.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
mentohust.c: In function 'gbk2utf':
mentohust.c:215:40: warning: passing argument 4 of 'gbk_to_utf8' from incompatible pointer type [-Wincompatible-pointer-types]
  gbk_to_utf8(gbksrc, gbklen, &utf8dst, &utf8len);
                                        ^~~~~~~~
In file included from mentohust.c:11:
strnormalize.h:28:83: note: expected 'unsigned int *' but argument is of type 'size_t *' {aka 'long unsigned int *'}
 int gbk_to_utf8(const char *from, unsigned int from_len, char **to, unsigned int *to_len);
                                                                     ~~~~~~~~~~~~~~^~~~~~
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c dlfunc.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include 
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c md5.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c mycheck.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c myconfig.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c myfunc.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
In file included from myfunc.c:25:
/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c strnormalize.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c myini.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include	
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c mystate.c -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1=mentohust-0.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -o mentohust mentohust.o dlfunc.o md5.o mycheck.o myconfig.o myfunc.o strnormalize.o myini.o mystate.o /home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/lib/libpcap.a -I/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/include -L/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/lib -L/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/lib -L/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/usr/lib -L/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/target-x86_64_musl/lib -L/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/runner/work/opde/opde/ctcgfw_sdk/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -ldl
x86_64-openwrt-linux-musl-gcc: error: /home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/libpcap-*/ipkg-install/usr/lib/libpcap.a: No such file or directory
Makefile:16: recipe for target 'mentohust' failed
make[4]: *** [mentohust] Error 1
make[4]: Leaving directory '/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1/src'
Makefile:59: recipe for target '/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1/.built' failed
make[3]: *** [/home/runner/work/opde/opde/ctcgfw_sdk/build_dir/target-x86_64_musl/mentohust-0.3.1/.built] Error 2
time: package/feeds/base/mentohust/compile#1.47#0.26#2.45

```

<details><summary>check-download.txt</summary>

```
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
[... cut ...]
CONFIG_PACKAGE_tvheadend=y
CONFIG_PACKAGE_verysync=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks_socks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR_socks=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Brook=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_kcptun=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_v2ray-plugin=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_simple-obfs=y

```
</details>

ElonH, 2020-04-23:
need test?

ElonH, 2020-04-23:
when the test is done, cherry-pick to `master` and `19.07` branch

AmadeusGhost, 2020-04-23:
> need test?

It seems that works fine.

commit 528a781c (2020-04-23): mentohust: more robust in makefile

fix #146
commit 7452cb0b (2020-04-28): mentohust: more robust in makefile

fix #146
closed 2020-04-23 by AmadeusGhost