immortalwrt/immortalwrt #150: ath10k要不要换成CT？

Aes64X, 2020-05-22:
我在竞斗云IPQ4019本身WIFI上截到：
只是偶见，由于是深夜，我并不知道会引起使用上的什么异常

[90778.793008] ------------[ cut here ]------------
[90778.793217] WARNING: CPU: 3 PID: 24 at backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x104/0x1ec [mac80211]
[90778.796749] STA ee:41:18:05:40:7a AC 2 txq pending airtime underflow: 4294967044, 252
[90778.796757] Modules linked in: xt_FULLCONENAT pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_string xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_bpf xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc oaf nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
[90778.865665]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat tcp_bbr cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs dummy rpcsec_gss_krb5 auth_rpcgss
[90778.937104]  oid_registry tun lockd sunrpc grace dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic sha1_arm_neon sha1_arm md5 ecb des_generic cts cbc authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 crc32c_generic
[90778.972248] CPU: 3 PID: 24 Comm: ksoftirqd/3 Not tainted 4.14.180 #0
[90778.994449] Hardware name: Generic DT based system
[90779.001073] [<c030e7a4>] (unwind_backtrace) from [<c030aa68>] (show_stack+0x10/0x14)
[90779.005665] [<c030aa68>] (show_stack) from [<c077faa4>] (dump_stack+0x94/0xa8)
[90779.013560] [<c077faa4>] (dump_stack) from [<c031f514>] (__warn+0xe0/0xfc)
[90779.020588] [<c031f514>] (__warn) from [<c031f188>] (warn_slowpath_fmt+0x38/0x48)
[90779.027554] [<c031f188>] (warn_slowpath_fmt) from [<bf71f990>] (ieee80211_sta_update_pending_airtime+0x104/0x1ec [mac80211])
[90779.035186] [<bf71f990>] (ieee80211_sta_update_pending_airtime [mac80211]) from [<bf71c090>] (ieee80211_tx_monitor+0xdf4/0xffc [mac80211])
[90779.046397] [<bf71c090>] (ieee80211_tx_monitor [mac80211]) from [<bf71c2e4>] (ieee80211_tx_status+0x4c/0x54 [mac80211])
[90779.058662] [<bf71c2e4>] (ieee80211_tx_status [mac80211]) from [<bf802534>] (ath10k_txrx_tx_unref+0x260/0x2e4 [ath10k_core])
[90779.069344] [<bf802534>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf7fdab0>] (ath10k_htt_txrx_compl_task+0xb00/0xfc8 [ath10k_core])
[90779.080708] [<bf7fdab0>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf847730>] (ath10k_pci_napi_poll+0x74/0x104 [ath10k_pci])
[90779.092626] [<bf847730>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c068f0b4>] (net_rx_action+0x138/0x2fc)
[90779.104623] [<c068f0b4>] (net_rx_action) from [<c0301520>] (__do_softirq+0xe0/0x244)
[90779.114080] [<c0301520>] (__do_softirq) from [<c032329c>] (run_ksoftirqd+0x30/0x50)
[90779.121982] [<c032329c>] (run_ksoftirqd) from [<c033dfdc>] (smpboot_thread_fn+0x198/0x1ac)
[90779.129360] [<c033dfdc>] (smpboot_thread_fn) from [<c033a7b4>] (kthread+0x148/0x154)
[90779.137693] [<c033a7b4>] (kthread) from [<c03076a8>] (ret_from_fork+0x14/0x2c)
[90779.145621] ---[ end trace 7f3949a90b11dccd ]---

Aes64X, 2020-05-22:
白天内网测速，竞斗云 QCA9984 时截到：
具体表现为测速大量丢包，吞吐极低只有10mb+左右，偶见个位数吞吐
ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Wed May 20 23:59:58 2020 kern.err kernel: [  254.425381] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:02 2020 kern.warn kernel: [  257.956348] ath10k_warn: 8 callbacks suppressed
Thu May 21 00:00:02 2020 kern.warn kernel: [  257.956363] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.err kernel: [  259.225379] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.494690] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.494800] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.500896] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:03 2020 kern.warn kernel: [  259.508238] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.515539] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.522762] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.530107] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.537372] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:04 2020 kern.warn kernel: [  259.544659] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:05 2020 kern.err kernel: [  260.575355] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.028366] ath10k_warn: 12 callbacks suppressed
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.028382] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:07 2020 kern.err kernel: [  263.465846] mmc0: Reset 0x1 never completed.
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.502654] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:07 2020 kern.warn kernel: [  263.502760] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.585660] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.585798] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.591915] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.599230] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  264.606510] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  265.440702] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:09 2020 kern.warn kernel: [  265.440829] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Thu May 21 00:00:11 2020 kern.err kernel: [  266.905424] mmc0: Reset 0x1 never completed.

AmadeusGhost, 2020-05-22:
这个有空我再测试看看

Aes64X, 2020-05-23:
> 这个我觉得其实换ct也解决不了什么，不过对于QCA9984来说换ct可能会好点
> 反正信道肯定是要固定一个而不是自动
> 
> ```
> ------------[ cut here ]------------
> WARNING: CPU: 3 PID: 24 xxx
> ```
> 
> 这种内核冲突的话你可以照着官方的commit升一下mac80211版本看看（rc就不升了），还有`mac80211.sh`别一起改
> 也有可能是这个 [commit](https://github.com/coolsnowwolf/lede/commit/1428e0eeed888727721cef16104574726ba48bd6) 影响，排除法也好做，内核换到4.19，就没有这些patch了
> 
> ```
> kern.err kernel: [ 263.465846] mmc0: Reset 0x1 never completed.
> ```
> 
> 这个去年12月的时候修过，现在的话我手头也没有sd卡，不知道什么情况
> mac80211确实4.19会稳不少，不过版本这个还是看仓库主怎么想吧，这里我也不是维护者
> 无线没有比较严重的问题的话我也不想管，一是无法复现，二是TG和issue天天被人骂/嘲讽，能正常说话的也没几个。上面说的您自己测测看吧，复现不了本菜鸡我无能为力。

么事么事，我自己处理吧。（Ath10K-CT 昨晚测试的竞斗云挂QCA9984 160MHz跑出了峰值710+的吞吐速率，反倒是IPQ4019本身给了8并发最高也就400+，不用CT的话IPQ4019本身WIFI能有560+）

Aes64X, 2020-05-25:
CT还是算了吧。。。。我实测了信号强度。。。实在是没法看。我看今天更新了MAC80211，稍后我会尝试下。

之前的测试，报错及大量丢包与4.14、4.19内核无关，与patch补丁无关。

AmadeusGhost, 2020-05-25:
@Aes64X 确实ct信号不怎么样

AmadeusGhost, 2020-05-25:
> CT还是算了吧。。。。我实测了信号强度。。。实在是没法看。我看今天更新了MAC80211，稍后我会尝试下。
> 
> 之前的测试，报错及大量丢包与4.14、4.19内核无关，与patch补丁无关。
> 
> 我跟了官方主线的MAC80211 后 ath10k 9984正常，但提示iw XXXX wifi无法启动，需要在停用启用一次才可以。

确实ct信号不大行，还有就是官方的mac80211.sh有点问题

Aes64X, 2020-05-25:
> > CT还是算了吧。。。。我实测了信号强度。。。实在是没法看。我看今天更新了MAC80211，稍后我会尝试下。
> > 之前的测试，报错及大量丢包与4.14、4.19内核无关，与patch补丁无关。
> > 我跟了官方主线的MAC80211 后 ath10k 9984正常，但提示iw XXXX wifi无法启动，需要在停用启用一次才可以。
> 
> 确实ct信号不大行，还有就是官方的mac80211.sh有点问题

炸了。。。。。

配置无线后页面有显示详细参数，终端无法搜索到信号。

关键日志：

[  551.595042] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
[  551.595203] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
[  558.402869] ath10k_pci 0000:01:00.0: unsupported HTC service id: 1536
[  558.605447] ath10k_pci 0000:01:00.0: Unknown eventid: 36933

QCA9984和IPQ4019本身无线都一样，可以配置，但无法搜索到信号。

AmadeusGhost, 2020-05-25:
> QCA9984和IPQ4019本身无线都一样，可以配置，但无法搜索到信号。

你是哪里的源码炸了？这里的话不应该啊，我前天试过正常

Aes64X, 2020-05-25:
> > QCA9984和IPQ4019本身无线都一样，可以配置，但无法搜索到信号。
> 
> 你是哪里的源码炸了？这里的话不应该啊，我前天试过正常

就这里的呀，最新pull......  我看你更了MAC80211就下手了。

Aes64X, 2020-05-25:
> > QCA9984和IPQ4019本身无线都一样，可以配置，但无法搜索到信号。
> 
> 你是哪里的源码炸了？这里的话不应该啊，我前天试过正常

内核可以排除了，我两台竞斗云，一台没挂9984，固件是昨晚更内核时编译的，没事，一切正常。
挂9984的是看你更80211了编译的，无线爆炸。

AmadeusGhost, 2020-05-25:
> 就这里的呀，最新pull...... 我看你更了MAC80211就下手了。

我这是正常的emmm
[image]
[image]

Aes64X, 2020-05-25:
> > 就这里的呀，最新pull...... 我看你更了MAC80211就下手了。
> 
> 我这是正常的emmm
> [image]
> [image]

MDZZ.......  真玄学 - -    

？？？你内核版本180？
我的181。。。。。

AmadeusGhost, 2020-05-25:
> 关键日志：
> 
> [ 551.595042] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
> [ 551.595203] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
> [ 558.402869] ath10k_pci 0000:01:00.0: unsupported HTC service id: 1536
> [ 558.605447] ath10k_pci 0000:01:00.0: Unknown eventid: 36933

这日志其实没啥用
> 你内核版本180？我的181

我前面说了我前天测的（编译刷入）

Aes64X, 2020-05-25:
> 这日志其实没啥用
>
>
> 我前面说了我前天测的（编译刷入）

稍等。。。。我上电发个完整日志

Aes64X, 2020-05-25:
> 这日志其实没啥用
>
>
> 我前面说了我前天测的（编译刷入）

哦哦哦，是这些。

Mon May 25 17:47:17 2020 daemon.notice netifd: radio1 (1367): command failed: Not supported (-95)
Mon May 25 17:47:17 2020 kern.info kernel: [   34.232314] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Mon May 25 17:47:17 2020 kern.info kernel: [   34.233247] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Mon May 25 17:47:17 2020 daemon.notice netifd: radio2 (1368): command failed: Not supported (-95)
Mon May 25 17:47:17 2020 daemon.notice netifd: radio0 (1365): command failed: Not supported (-95)

cncherisher, 2020-05-25:
我刚刚编译的acrh17 5G也挂了，可以配置，但无法搜索到信号。过一段时间路由死机重启
附日志
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.417909] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.418010] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.424140] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.431465] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.438717] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.445997] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.453252] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.460593] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.467903] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:09 2020 kern.warn kernel: [  394.475139] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:33 2020 kern.warn kernel: [  413.045815] ath10k_warn: 19 callbacks suppressed
Mon May 25 20:49:33 2020 kern.warn kernel: [  413.045854] ath10k_pci 0000:01:00.0: wmi command 36967 timeout, restarting hardware
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.086666] ath10k_pci 0000:01:00.0: failed to send pdev bss chan info request
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.091524] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.093172] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.101501] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.107469] ath10k_pci 0000:01:00.0: failed to set cts protection for vdev 0: -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.115786] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  416.123140] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -108
Mon May 25 20:49:33 2020 kern.warn kernel: [  417.906776] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:33 2020 kern.warn kernel: [  417.907022] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon May 25 20:49:33 2020 kern.info kernel: [  418.020367] ieee80211 phy0: Hardware restart was requested
Mon May 25 20:49:39 2020 kern.warn kernel: [  423.959768] ath10k_warn: 34 callbacks suppressed
Mon May 25 20:49:39 2020 kern.warn kernel: [  423.960378] ath10k_pci 0000:01:00.0: unsupported HTC service id: 1536
Mon May 25 20:49:39 2020 kern.warn kernel: [  424.177765] ath10k_pci 0000:01:00.0: Unknown eventid: 36933
Mon May 25 20:49:55 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon May 25 20:49:55 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Mon May 25 20:49:55 2020 kern.warn kernel: [  440.354430] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
Mon May 25 20:49:55 2020 kern.warn kernel: [  440.354777] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
Mon May 25 20:49:56 2020 daemon.notice netifd: radio0 (7323): command failed: Not supported (-95)
Mon May 25 20:55:45 2020 daemon.notice netifd: radio1 (10282): command failed: Not supported (-95)

AmadeusGhost, 2020-05-25:
> 我刚刚编译的acrh17 5G也挂了，可以配置，但无法搜索到信号。过一段时间路由死机重启

acrh17需要一个懂代码并且有这台机子的来调试，没记错的话17自从mac80211上了5.4无线就没好过

AmadeusGhost, 2020-05-26:
> 你内核版本180？我的181
 
我拿最新源码+mac80211升到5.4.36，一切正常

Aes64X, 2020-05-26:
> > 你内核版本180？我的181
> 
> 我拿最新源码+mac80211升到5.4.36，一切正常

我删掉整个代码目录重新拉再试试。。。。

Aes64X, 2020-05-26:
> > 你内核版本180？我的181
> 
> 我拿最新源码+mac80211升到5.4.36，一切正常

算了，直接合吧。。。有问题我切CT，反正就小房间跑个160MHz，无伤大雅 --

closed 2020-05-26 by Aes64X