immortalwrt/immortalwrt #154: hiwif-c526a编译报错 错误如下

alphavip, 2020-06-02:
touch /home/alpha/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/mt76-2020-03-10-08054d5a/.prepared_e0dceebb6c43a7ced55f13007ec0419e_6664517399ebbbc92a37c5bb081b5c53_check
. /home/alpha/openwrt/include/shell.sh; xzcat /home/alpha/openwrt/dl/mt76-2020-03-10-08054d5a.tar.xz | tar -C /home/alpha/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/mt76-2020-03-10-08054d5a/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/alpha/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/mt76-2020-03-10-08054d5a

Applying ./patches/001-allow-vht-on-2g.patch using plaintext: 
patching file mac80211.c
Hunk #1 FAILED at 156.
Hunk #2 FAILED at 218.
Hunk #3 succeeded at 349 with fuzz 1 (offset -96 lines).
2 out of 3 hunks FAILED -- saving rejects to file mac80211.c.rej
Patch failed!  Please fix ./patches/001-allow-vht-on-2g.patch!
Makefile:257: recipe for target '/home/alpha/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/mt76-2020-03-10-08054d5a/.prepared_e0dceebb6c43a7ced55f13007ec0419e_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/alpha/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/mt76-2020-03-10-08054d5a/.prepared_e0dceebb6c43a7ced55f13007ec0419e_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/alpha/openwrt/package/kernel/mt76'

经过对比发现/home/alpha/openwrt/package/kernel/mt76/pacthes有问题

alphavip, 2020-06-02:
diff --git a/package/kernel/mt76/patches/001-allow-vht-on-2g.patch b/package/kernel/mt76/patches/001-allow-vht-on-2g.patch
index 571574980b..bd8cf789a2 100644
--- a/package/kernel/mt76/patches/001-allow-vht-on-2g.patch
+++ b/package/kernel/mt76/patches/001-allow-vht-on-2g.patch
@@ -4,10 +4,10 @@
  void mt76_set_stream_caps(struct mt76_dev *dev, bool vht)
  {
        if (dev->cap.has_2ghz)
--              mt76_init_stream_cap(dev, &dev->phy.sband_2g.sband, false);
-+              mt76_init_stream_cap(dev, &dev->phy.sband_2g.sband, vht);
+-              mt76_init_stream_cap(dev, &dev->sband_2g.sband, false);
++              mt76_init_stream_cap(dev, &dev->sband_2g.sband, vht);
        if (dev->cap.has_5ghz)
-               mt76_init_stream_cap(dev, &dev->phy.sband_5g.sband, vht);
+               mt76_init_stream_cap(dev, &dev->sband_5g.sband, vht);
  }
 @@ -218,14 +218,14 @@ mt76_init_sband(struct mt76_dev *dev, st
  
@@ -16,9 +16,9 @@
 -                 int n_rates)
 +                 int n_rates, bool vht)
  {
-       dev->hw->wiphy->bands[NL80211_BAND_2GHZ] = &dev->phy.sband_2g.sband;
+       dev->hw->wiphy->bands[NL80211_BAND_2GHZ] = &dev->sband_2g.sband;
  
-       return mt76_init_sband(dev, &dev->phy.sband_2g,
+       return mt76_init_sband(dev, &dev->sband_2g,
                               mt76_channels_2ghz,
                               ARRAY_SIZE(mt76_channels_2ghz),
 -                             rates, n_rates, false);

AmadeusGhost, 2020-06-02:
@alphavip Already fixed in yesterday, and I forgot to push to default branch, just run git pull.

closed 2020-06-02 by alphavip