immortalwrt/immortalwrt #156: openwrt-18.06编译一直不成功

kenzok8, 2020-06-05:
openwrt-18.06编译一直不成功

ElonH, 2020-06-05:
麻烦先阅读以下文章
https://www.chiark.greenend.org.uk/~sgtatham/bugs-cn.html

kenzok8, 2020-06-05:
1、github用云编译出错
2、本地物理机编译也出错
3、连续出错情况已经5天以上，4.19.124时一直能正常编译，
4、本地物理机默认编译也出错
ubuntu 18.04

kenzok8, 2020-06-05:
https://github.com/kenzok8/LEDE-x86_64/runs/741477512?check_suite_focus=true#step:9:48588
cp -fpR /home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1/ipkg-install/* /home/k
find /home/kenzo/ctc/openwrt/tmp/stage-c-ares -name '*.la' | xargs -r rm -f;
if [ -f /home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/packages/c-ares.list ]; then /homeaging_dir/target-x86_64_musl/packages/c-ares.list" "/home/kenzo/ctc/openwrt/staging_dir/target-x8
if [ -d /home/kenzo/ctc/openwrt/tmp/stage-c-ares ]; then (cd /home/kenzo/ctc/openwrt/tmp/stage-c-HELL= flock /home/kenzo/ctc/openwrt/tmp/.staging-dir.flock -c ' mv /home/kenzo/ctc/openwrt/tmp/st/packages/c-ares.list && cp -fpR /home/kenzo/ctc/openwrt/tmp/stage-c-ares/* /home/kenzo/ctc/openw
rm -rf /home/kenzo/ctc/openwrt/tmp/stage-c-ares
touch /home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.c-ares_installed
make[3]: Leaving directory '/home/kenzo/ctc/openwrt/feeds/packages/libs/c-ares'
time: package/feeds/packages/c-ares/compile#71.89#12.32#92.03
make[2]: Leaving directory '/home/kenzo/ctc/openwrt'
package/Makefile:107: recipe for target '/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/s
make[1]: *** [/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Erro
make[1]: Leaving directory '/home/kenzo/ctc/openwrt'
/home/kenzo/ctc/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

AmadeusGhost, 2020-06-06:
你的问题是选上了 bcm27xx-userland 这个包，实际上这个包 `make menuconfig` 是树莓派才能选的，不清楚你是怎么选的。还有就是发 log 请找到具体 error 的地方，你发的这个 log 就相当于没发，我还得自己去找，大家的时间都很宝贵。
```
2020-06-05T09:34:36.3369949Z make[6]: *** [interface/khronos/CMakeFiles/khrn_client.dir/common/khrn_int_hash_asm.s.o] Error 1
2020-06-05T09:34:36.3370074Z make[5]: *** [interface/khronos/CMakeFiles/khrn_client.dir/all] Error 2
2020-06-05T09:34:36.3370188Z make[4]: *** [all] Error 2
2020-06-05T09:34:36.3370584Z make[3]: *** [/home/runner/work/LEDE-x86_64/LEDE-x86_64/openwrt/build_dir/target-x86_64_musl/bcm27xx-userland-f4bccc38f9d0a740aeb7818d33ca1fd551e325de/.built] Error 2
2020-06-05T09:34:36.3370892Z time: package/utils/bcm27xx-userland/compile#4.93#0.87#6.66
2020-06-05T09:34:36.3371216Z make[2]: *** [package/utils/bcm27xx-userland/compile] Error 2
2020-06-05T09:34:36.3371331Z make[2]: *** Waiting for unfinished jobs....
2020-06-05T09:34:36.3371632Z package/Makefile:111: recipe for target 'package/utils/bcm27xx-userland/compile' failed
```

kenzok8, 2020-06-06:
谢谢回复，但是问题是我重新下载源码编译，多次编译，包括云编译。默认config也是这个错误呢，所以才反馈

kenzok8, 2020-06-06:
Symbol: PACKAGE_bcm27xx-userland [=n]

        x
  x Type  : tristate

             x
  x Defined at tmp/.config-package.in:135923

             x
  x   Prompt: bcm27xx-userland. BCM27xx userland tools

             x
  x   Depends on: TARGET_bcm27xx [=n]

            x
  x   Location:

            x
  x (1) -> Utilities

             x
  x Selects: PACKAGE_librt [=y] && PACKAGE_libpthread [=y] && PACKAGE_libc
[=y] && PACKAGE_libssp [=n]
             x
  x Selected by [n]:

             x
  x   - PACKAGE_autocore-raspberrypi [=n] && TARGET_bcm27xx [=n]

kenzok8, 2020-06-06:
make[3]: Leaving directory
'/home/kenzo/ctc/openwrt/feeds/packages/libs/c-ares'
time: package/feeds/packages/c-ares/compile#73.55#12.69#93.23
make[2]: Leaving directory '/home/kenzo/ctc/openwrt'
package/Makefile:107: recipe for target
'/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile'
failed
make[1]: ***
[/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile]
Error 2
make[1]: Leaving directory '/home/kenzo/ctc/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/kenzo/ctc/openwrt/include/toplevel.mk:222: recipe for target 'world'
failed
make: *** [world] Error 1

kenzok8, 2020-06-06:
make[6]: Leaving directory
'/home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1'
make[5]: Leaving directory
'/home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1'
make[4]: Leaving directory
'/home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1'
touch
/home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1/.built
rm -rf /home/kenzo/ctc/openwrt/tmp/stage-c-ares
mkdir -p /home/kenzo/ctc/openwrt/tmp/stage-c-ares/host
/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/packages
/home/kenzo/ctc/openwrt/staging_dir/host/packages
install -d -m0755 /home/kenzo/ctc/openwrt/tmp/stage-c-ares
cp -fpR
/home/kenzo/ctc/openwrt/build_dir/target-x86_64_musl/c-ares-1.16.1/ipkg-install/*
/home/kenzo/ctc/openwrt/tmp/stage-c-ares/
find /home/kenzo/ctc/openwrt/tmp/stage-c-ares -name '*.la' | xargs -r rm -f;
if [ -f
/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/packages/c-ares.list
]; then /home/kenzo/ctc/openwrt/scripts/clean-package.sh
"/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/packages/c-ares.list"
"/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl"; fi
if [ -d /home/kenzo/ctc/openwrt/tmp/stage-c-ares ]; then (cd
/home/kenzo/ctc/openwrt/tmp/stage-c-ares; find ./ >
/home/kenzo/ctc/openwrt/tmp/stage-c-ares.files);      SHELL= flock
/home/kenzo/ctc/openwrt/tmp/.staging-dir.flock -c ' mv
/home/kenzo/ctc/openwrt/tmp/stage-c-ares.files
/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/packages/c-ares.list
&& cp -fpR /home/kenzo/ctc/openwrt/tmp/stage-c-ares/*
/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/kenzo/ctc/openwrt/tmp/stage-c-ares
touch
/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.c-ares_installed
make[3]: Leaving directory
'/home/kenzo/ctc/openwrt/feeds/packages/libs/c-ares'
time: package/feeds/packages/c-ares/compile#64.87#10.56#81.50
make[2]: Leaving directory '/home/kenzo/ctc/openwrt'
package/Makefile:107: recipe for target
'/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile'
failed
make[1]: ***
[/home/kenzo/ctc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile]
Error 2
make[1]: Leaving directory '/home/kenzo/ctc/openwrt'

1715173329, 2020-06-06:
Also appear to https://github.com/project-openwrt/OpenWrt-Compilation-Checker/actions/runs/124904099, try to build without autocore-x86.

ledewrt, 2020-06-06:
我也是一直无法编译。

kenzok8, 2020-06-06:
谢谢分享！

AmadeusGhost, 2020-06-06:
`git pull`然后`rm .config`重新选择应该就行了

closed 2020-06-06 by AmadeusGhost

kenzok8, 2020-06-06:
还真不行，楼上 rm -Rf package/utils/bcm27xx-userland才是正解

源码重新下载默认编译都是这个错误，不是我多选择了什么，而且我云编译也是这个错误，默认config

连这个git clean -xdf命令也用了，dl文件包也删除过

AmadeusGhost, 2020-06-06:
> 还真不行，楼上 rm -Rf package/utils/bcm27xx-userland才是正解

`make package/bcm27xx-userland/clean`或者直接`make clean`