immortalwrt/immortalwrt #157: [建议]增加 gowedav 多方案配置

imoldchen, 2020-06-07:
webdav 目前只能配置一个共享方案, 之前用的 lighttpd+webdav 的方式, 可以配置一个端口下多用户并对应不同的目录, 因为路由器挂载了两个硬盘,作为私有云盘, 分别两人使用, 看到 gowebdav 已经有配置界面了, 希望配置界面能实现添加多个方案的功能.

ElonH, 2020-06-07:
https://github.com/hacdias/webdav

这个可以满足你的需求，不过并没做openwrt的适配。

imoldchen, 2020-06-07:
> https://github.com/hacdias/webdav
> 
> 这个可以满足你的需求，不过并没做openwrt的适配。

用这个好像在 OpenWRT 无法运行, 我还是用 caddy+webdav 吧, 看 gowebdav 以后能不能增加了, 现在的只能一个目录,确实不好用.

1715173329, 2020-06-07:
@imoldchen gowebdav 只支持单用户

1715173329, 2020-06-20:
> https://github.com/hacdias/webdav
> 
> 这个可以满足你的需求，不过并没做openwrt的适配。

We just need write a LuCI.

ElonH, 2020-06-20:
@1715173329 
> We just need to write a LuCI.

I'm affair with the compilation of golang. Some golang compile failure in OpenWRT.

Another problem, but not affect your suggestion,
 I have used this program for a while in ubuntu, some of files and directories are not shown in this program, and I still don't have any clue. I don't know if my mistaken in configuration raises up this problem.

But it's not a bad idea to use `hacdias/webdav` with Luci. Need a hand? I can try to compile it and test it.