immortalwrt/immortalwrt #158: 升级内核后，BaiduPCS导致编译失败。

ledewrt, 2020-06-15:
报错如下：
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/dl/BaiduPCS-Go-v3.7.1-linux-amd64.zip] Error 2
Makefile:88: recipe for target '/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/dl/BaiduPCS-Go-v3.7.1-linux-amd64.zip' failed
make[3]: Leaving directory '/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/package/lean/baidupcs-web'
time: package/lean/baidupcs-web/compile#0.21#0.05#1.56
package/Makefile:111: recipe for target 'package/lean/baidupcs-web/compile' failed
make[2]: *** [package/lean/baidupcs-web/compile] Error 2
make[2]: Leaving directory '/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede'
make[1]: *** [/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede'
/home/runner/work/lede-1806-ctcgfw/lede-1806-ctcgfw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.

AmadeusGhost, 2020-06-15:
跟升级内核没关系，是作者删库了

ledewrt, 2020-06-15:
哦，知道了。多谢

closed 2020-06-20 by 1715173329