immortalwrt/immortalwrt #161: 请问comfast cf-811ac usb无线网卡x86 openwrt能驱动吗？

yu13740000, 2020-06-21:
请问comfast cf-811ac usb无线网卡x86 openwrt能驱动吗？这个网卡在r2s上能驱动，好像8811cu和8821cu同驱动，x86 openwrt软路由上试了你的8821cu没成功，不知道是不是姿势不对，还是不支持这个网卡。麻烦大佬看看，谢谢！
就是链接这块网卡： https://detail.tmall.com/item.htm?spm=a230r.1.14.58.36dd3f5eTX0Sa6&id=591582535513&ns=1&abbucket=8&skuId=4059416376519

AmadeusGhost, 2020-06-21:
不能用那具体日志呢？还有就是我们并没有相关的硬件。

yu13740000, 2020-06-21:
日志现在看不到了，固件换了，折腾了2天没成功

AmadeusGhost, 2020-06-25:
我查了下这个网卡 linux 下就是用的 https://github.com/brektrou/rtl8821CU 这个源码，如果不能提供日志的话请自行 close issue

yu13740000, 2020-06-26:
Fri Jun 26 09:52:17 2020 kern.info kernel: [    1.458132] Button Hotplug driver version 0.4.1
Fri Jun 26 09:52:17 2020 user.info kernel: [    1.458352] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Fri Jun 26 09:52:17 2020 user.info kernel: [    1.468363] init: - preinit -
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    1.569500] random: jshn: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    1.583244] random: jshn: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    1.584699] random: jshn: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 kern.warn kernel: [    4.613477] urandom_read: 4 callbacks suppressed
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.613479] random: jshn: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.638973] random: procd: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.775475] F2FS-fs (loop0): recover_inode: ino = 212, name = .shadowsocksr.uci-mJgIil, inline = b
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.777712] F2FS-fs (loop0): recover_dentry: ino = 212, name = .shadowsocksr.uci-mJgIil, dir = e, err = 0
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.778022] F2FS-fs (loop0): recover_data: ino = 212 (i_size: recover) recovered = 0, err = 0
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.781997] F2FS-fs (loop0): checkpoint: version = 178eb896
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    4.813224] F2FS-fs (loop0): Mounted with checkpoint version = 178eb896
Fri Jun 26 09:52:17 2020 user.info kernel: [    4.813752] mount_root: switching to f2fs overlay
Fri Jun 26 09:52:17 2020 kern.warn kernel: [    4.832018] EXT4-fs (sda1): warning: mounting unchecked fs, running e2fsck is recommended
Fri Jun 26 09:52:17 2020 kern.info kernel: [    4.837947] EXT4-fs (sda1): mounted filesystem without journal. Opts: (null)
Fri Jun 26 09:52:17 2020 user.warn kernel: [    4.839072] urandom-seed: Seeding with /etc/urandom.seed
Fri Jun 26 09:52:17 2020 user.info kernel: [    4.858505] procd: - early -
Fri Jun 26 09:52:17 2020 user.info kernel: [    5.445003] procd: - ubus -
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    5.454759] random: ubusd: uninitialized urandom read (4 bytes read)
Fri Jun 26 09:52:17 2020 user.info kernel: [    5.495601] procd: - init -
Fri Jun 26 09:52:17 2020 user.info kernel: [    5.580580] urngd: v1.0.2 started.
Fri Jun 26 09:52:17 2020 kern.notice kernel: [    5.589198] random: crng init done
Fri Jun 26 09:52:17 2020 user.info kernel: [    5.606437] kmodloader: loading kernel modules from /etc/modules.d/*
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.607810] NET: Registered protocol family 38
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.609040] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.611278] e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.611485] e1000: Copyright (c) 1999-2006 Intel Corporation.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.612107] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.4.0-k
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.612313] igb: Copyright (c) 2007-2014 Intel Corporation.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.612709] igbvf: Intel(R) Gigabit Virtual Function Network Driver - version 2.4.0-k
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.612982] igbvf: Copyright (c) 2009 - 2012 Intel Corporation.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.614179] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver - version 5.1.0-k
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.614452] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.615012] ehci-platform: EHCI generic platform driver
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.622545] cryptodev: driver 1.10 loaded.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.622937] i2c /dev entries driver
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.624108] 8139too: 8139too Fast Ethernet driver 0.9.28
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.625749] usbcore: registered new interface driver cdc_wdm
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.626186] Loading modules backported from Linux version v4.19.98-0-gd183c8e2647a
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.626454] Backport generated by backports.git v4.19.98-1-0-g8204eb99
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.627247] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.627437] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.627790] e1000e 0000:00:19.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.628096] e1000e 0000:00:19.0: Interrupt Mode set to 1
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.738601] e1000e 0000:00:19.0 0000:00:19.0 (uninitialized): registered PHC clock
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.833255] e1000e 0000:00:19.0 eth0: (PCI Express:2.5GT/s:Width x1) 00:23:24:80:b5:c0
Fri Jun 26 09:52:17 2020 kern.info kernel: [    5.833588] e1000e 0000:00:19.0 eth0: Intel(R) PRO/1000 Network Connection
[... cut ...]
Fri Jun 26 09:52:27 2020 daemon.info pppd[2365]: Renamed interface ppp0 to pppoe-wan
Fri Jun 26 09:52:27 2020 daemon.info pppd[2365]: Using interface pppoe-wan
Fri Jun 26 09:52:27 2020 daemon.notice pppd[2365]: Connect: pppoe-wan <--> eth1
Fri Jun 26 09:52:27 2020 daemon.info pppd[2365]: syncppp not active
Fri Jun 26 09:52:27 2020 daemon.info pppd[2365]: CHAP authentication succeeded: Authentication success,Welcome!
Fri Jun 26 09:52:27 2020 daemon.notice pppd[2365]: CHAP authentication succeeded

AmadeusGhost, 2020-06-26:
先插上网卡再接上电源能正常工作嘛
Someone said that, "I am observing a recurrent problem that I have seen before in old posts on the Internet. The issue is related to warm boot. Driver works well when I plug my USB device (rtl8821cu) in USB port when the system is turned on, also when the system does a cold boot. But if I perform a warm boot with the USB device plugged in, then the problems appear."

yu13740000, 2020-06-26:
这就是插上启动的日记。无论启动前后插都不能工作

1715173329, 2020-07-10:
We do not have the device, so you may need solve it by yourself.

closed 2020-07-10 by 1715173329

AmadeusGhost, 2020-08-15:
现在能驱动，不过因为垃圾 Realtek 不支持 WEXT API 和 openwrt 的 nl80211，无法通过 luci 配置