immortalwrt/immortalwrt #170: 问题反馈AdGuardHome

fuqiang03, 2020-07-20:
define Package/AdGuardHome/install
	$(call GoPackage/Package/Install/Bin,$(PKG_INSTALL_DIR))
	$(INSTALL_DIR) $(1)/usr/bin/AdGuardHome
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/AdGuardHome $(1)/usr/bin/AdGuardHome/AdGuardHome

和配置文件路径保持一致

虽然客户端执行了,但是无法打开页面.
[image]
[image]

closed 2020-07-20 by 1715173329

fuqiang03, 2020-07-20:
使用源码作者的客户端没有这些问题
我只是想说可能在AdGuardHome 编译的时候缺少了什么东西
[image]
[image]

fuqiang03, 2020-07-27:
@1715173329 
[image]
请问有GitHub上的资源可以替代吗的(go.mod go.sum),或者其他的办法,谢谢
由于无法翻墙所以造成了上面的错误

1715173329, 2020-07-27:
设置个 GOPROXY 应该就行，你可以谷歌看看

fuqiang03, 2020-07-27:
ifeq ($(CONFIG_AdGuardHome_COMPRESS_GOPROXY),y)
export GO111MODULE=on
export GOPROXY=https://goproxy.cn
endif
这样的地址可以吗

fuqiang03, 2020-08-01:
@1715173329 加了代理也是无效的 改了go.mod go.sum文件也不行
[image]
[image]

1715173329, 2020-08-01:
那就加钱换梯子

QiuSimons, 2020-08-01:
> @1715173329
> [image]
> 请问有GitHub上的资源可以替代吗的(go.mod go.sum),或者其他的办法,谢谢
> 由于无法翻墙所以造成了上面的错误

用action跑呗，墙外环境编译（doge

fuqiang03, 2020-08-01:
没有梯子真的不方便 其他的都可以用代理 只有这个不行 不知道问题出在哪里