immortalwrt/immortalwrt #172: openwrt-19.07编译刷入后打开turboacc报错

chuansao-258, 2020-07-21:
信息如下：
/usr/lib/lua/luci/model/cbi/turboacc.lua:1: module 'luci.model.ipkg' not found:
	no field package.preload['luci.model.ipkg']
	no file './luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg/init.lua'
	no file '/usr/lib/lua/luci/model/ipkg.lua'
	no file '/usr/lib/lua/luci/model/ipkg/init.lua'
	no file './luci/model/ipkg.so'
	no file '/usr/lib/lua/luci/model/ipkg.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/turboacc.lua:1: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1340: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1023: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

chuansao-258, 2020-07-21:
而且没有DHCP。手动改IP后才进入管理页面

1715173329, 2020-07-21:
> 而且没有DHCP。手动改IP后才进入管理页面

上个图看看
关于 DHCP 设置页面

chuansao-258, 2020-07-21:
太折腾，不想重刷了/(ㄒoㄒ)/~~

commit a57fb98e (2020-07-21): include/target: add some necessary package

Fix #172
closed 2020-07-21 by 1715173329