immortalwrt/immortalwrt #188: 可否给Openwrt-19.07加上efi引导呢

RayorRui, 2020-08-31:


AmadeusGhost, 2020-09-08:
人呢...？😮

Teaffanie, 2020-09-08:
看到了，感谢dalao

AmadeusGhost, 2020-09-08:
我萌新， https://github.com/project-openwrt/openwrt/commit/ced7cd5d911a27e9247dc329376e6668b4c69f69 Can you test this?

Teaffanie, 2020-09-08:
> 我萌新， ced7cd5 Can you test this?

开始编译了

Teaffanie, 2020-09-08:
> 我萌新， [ced7cd5](https://github.com/project-openwrt/openwrt/commit/ced7cd5d911a27e9247dc329376e6668b4c69f69) Can you test this?

报错了。
grub-mkimage: error: cannot open `/workdir/openwrt/staging_dir/host/lib/grub/i386-pc/moddep.lst': No such file or directory.
2779
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/bzImage /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/root.grub/boot/vmlinuz
2780
make[5]: *** [install-images] Error 1

AmadeusGhost, 2020-09-08:
> 报错了。
> make[5]: *** [install-images] Error 1
> https://github.com/Teaffanie/bygj/runs/1085478096?check_suite_focus=true

现在应该可以了 (just my mistake)

Teaffanie, 2020-09-09:
> 现在应该可以了 (just my mistake)
编译成功并且正常使用了

closed 2020-09-09 by 1715173329