immortalwrt/immortalwrt #189: master 分支的 filebrowser，在 luci 不能启动

fanck0605, 2020-08-31:
[image]

dalao，master 分支的 filebrowser 在 luci 不能启动。
手动执行 `/etc/init.d/filebrowser reload` 才行。
在脚本里加了点标记，在 luci 选择已启用时，`/etc/init.d/filebrowser` 的确没被执行
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2019 [CTCGFW] Project OpenWRT

echo 0 >> /fb_test

. /lib/functions.sh
. /lib/functions/procd.sh

USE_PROCD=1

START=90
STOP=10

init_conf() {
        echo 1 >> /fb_test
        config_load "filebrowser"
        config_get "addr_type" "config" "addr_type" "lan"
        ...
```

lunatickochiya, 2020-12-19:
这是filebrowser查找路径的BUG  系统日志里可以看到

1715173329, 2020-12-19:
> 这是filebrowser查找路径的BUG 系统日志里可以看到

请问能否提供相关日志？

lunatickochiya, 2020-12-19:
> > 这是filebrowser查找路径的BUG 系统日志里可以看到
> 
> 请问能否提供相关日志？

Sat Dec 19 16:16:37 2020 daemon.err uhttpd[1823]: 2020/12/19 08:16:37 exec: "getent": executable file not found in $PATH
Sat Dec 19 16:16:40 2020 daemon.err uhttpd[1823]: 2020/12/19 08:16:40 exec: "getent": executable file not found in $PATH

zxlhhyccc, 2020-12-22:
> > > 这是filebrowser查找路径的BUG 系统日志里可以看到
> > 
> > 
> > 请问能否提供相关日志？
> 
> Sat Dec 19 16:16:37 2020 daemon.err uhttpd[1823]: 2020/12/19 08:16:37 exec: "getent": executable file not found in $PATH
> Sat Dec 19 16:16:40 2020 daemon.err uhttpd[1823]: 2020/12/19 08:16:40 exec: "getent": executable file not found in $PATH

filebrowser在master下能启动吧，你是不是用错了luci？

lunatickochiya, 2020-12-22:
> filebrowser在master下能启动吧，你是不是用错了luci？

不是，不能启动，我用的确实不是master版本，但是我可以肯定你的master版本不能在luci上关闭filebrowser后再在luci启动

zxlhhyccc, 2021-01-02:
> [image]
> 
> dalao，master 分支的 filebrowser 在 luci 不能启动。
> 手动执行 `/etc/init.d/filebrowser reload` 才行。
> 在脚本里加了点标记，在 luci 选择已启用时，`/etc/init.d/filebrowser` 的确没被执行
> 
> ```
> #!/bin/sh /etc/rc.common
> # Copyright (C) 2019 [CTCGFW] Project OpenWRT
> 
> echo "exec_script" >> /fb_test
> 
> . /lib/functions.sh
> . /lib/functions/procd.sh
> 
> USE_PROCD=1
> 
> START=90
> STOP=10
> 
> init_conf() {
>         echo "init_conf" >> /fb_test
>         config_load "filebrowser"
>         config_get "addr_type" "config" "addr_type" "lan"
>         ...
> ```
今天编译了filebrowser，在master下确实不能启动。。。。

Lienol, 2021-01-08:
似曾相识的BUG，我之前似乎经历过。忘了怎样解决了。不过我不用这个插件了。
似乎是加了[这个](https://github.com/Lienol/openwrt-package/blob/87bec4ddbf4eccbd4342e6efab3a54ddeb74d16a/luci-app-filebrowser/root/etc/init.d/filebrowser#L25)
可以尝试下

1715173329, 2021-01-08:
> 似曾相识的BUG，我之前似乎经历过。忘了怎样解决了。不过我不用这个插件了。
> 似乎是加了[这个](https://github.com/Lienol/openwrt-package/blob/87bec4ddbf4eccbd4342e6efab3a54ddeb74d16a/luci-app-filebrowser/root/etc/init.d/filebrowser#L25)
> 可以尝试下

现在只有做一个 workaround 来“解决”这个问题： 050e950eee

这个问题似乎与 HOME 环境变量并无关系
问题来自于 PROCD 守护，即使加入了 `reload_trigger`，在配置文件被更改后，procd 也不会使其被重载
并且在 ucitrack 中绑定 init 后，在 LuCI 更新配置文件也不会自动调用 init 重载
但如果不使用 procd，则 init 调用正常

但很有意思的是目前可以使其被停止，却无法启动。

lunatickochiya, 2021-01-09:
> 似曾相识的BUG，我之前似乎经历过。忘了怎样解决了。不过我不用这个插件了。
> 似乎是加了[这个](https://github.com/Lienol/openwrt-package/blob/87bec4ddbf4eccbd4342e6efab3a54ddeb74d16a/luci-app-filebrowser/root/etc/init.d/filebrowser#L25)
> 可以尝试下

加了 依然无法启用

Lienol, 2021-01-09:
[image]
我用我的魔改master分支试了一下似乎并没有问题，用的是https://github.com/project-openwrt/openwrt/commit/050e950eee24401bf35328b14de5c1f9e71e6b37 之前的

1715173329, 2021-01-09:
> 我用我的魔改master分支试了一下似乎并没有问题

应该是和您的这个提交有关： Lienol/openwrt-luci@2b1c32a25f511710e9254301b8099efa10ed04f3
最新的 LuCI 有一些行为应该被更改了

BTW，我最近发现了一个新的 procd 触发写法，不过暂不确定是否工作
```bash
procd_add_config_trigger "config.change" "filebrowser" /etc/init.d/filebrowser reload
```

lunatickochiya, 2021-01-09:
btw 我用的lede 17.01 luci  我更希望楼上的方法有效

ksong008, 2021-01-24:
已经正常了，关闭

lunatickochiya, 2021-01-24:
> 已经正常了，可以关闭了

哪里更新了？

ksong008, 2021-01-24:
> > 已经正常了，可以关闭了
> 
> 哪里更新了？

我昨天编译了master，filebrowser能正常工作

lunatickochiya, 2021-01-24:
> > > 已经正常了，可以关闭了
> > 
> > 
> > 哪里更新了？
> 
> 我昨天编译了master，filebrowser能正常工作

能在luci关闭后再在luci开启？

ksong008, 2021-01-24:
@lunatickochiya 我确定。。。

1715173329, 2021-01-24:
现在的只是一个 hack，procd 的问题并没有解决
等我考试完了再来测试

fanck0605, 2021-01-24:
@lunatickochiya 这两天编译的版本的确又没问题了，玄学bug

1715173329, 2021-01-31:
https://openwrt.org/docs/guide-developer/procd-init-scripts#debugging
Maybe we need `PROCD_DEBUG=1` and `INIT_TRACE=1` to see what had fucking happened.