immortalwrt/immortalwrt #195: Can't compile after disable ipv6 support

gothdawid, 2020-09-10:
[config2.txt](https://github.com/project-openwrt/openwrt/files/5201185/config2.txt)

`touch /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/stamp/.luci-i18n-tinyproxy-pl_installed
echo "luci-i18n-tinyproxy-pl" >> /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/luci-app-tinyproxy.default.install
mkdir -p /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/stamp
SHELL= flock /home/studzik/CTCGFW/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/luci-app-tinyproxy/.pkgdir/luci-i18n-tinyproxy-zh-cn/. /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/'
touch /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/stamp/.luci-i18n-tinyproxy-zh-cn_installed
echo "luci-i18n-tinyproxy-zh-cn" >> /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/luci-app-tinyproxy.default.install
make[3]: Leaving directory '/home/studzik/CTCGFW/openwrt/feeds/luci/applications/luci-app-tinyproxy'
time: package/feeds/luci/luci-app-tinyproxy/compile#1.07#0.32#1.44
make[3]: Entering directory '/home/studzik/CTCGFW/openwrt/feeds/packages/net/miniupnpd'
rm -f /home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510/.built
touch /home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510:miniupnpd-2.1.20200510 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto  -I/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510:miniupnpd-2.1.20200510 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto  -I/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include " LDFLAGS="-L/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed " make  -C /home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510:miniupnpd-2.1.20200510 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -flto" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510'
mips-openwrt-linux-musl-gcc -L/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed  -L/home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/usr/lib  testupnpdescgen.o upnpdescgen.o  -lip4tc -lip6tc -lmnl -lnetfilter_conntrack -lnfnetlink -lssl -lcrypto -ldl -luuid  -o testupnpdescgen
/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: /home/studzik/CTCGFW/openwrt/tmp/ccpq7uwQ.ltrans0.ltrans.o: in function `main':
<artificial>:(.text.startup+0x3ec): undefined reference to `gen6FC'
/home/studzik/CTCGFW/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld: <artificial>:(.text.startup+0x4d8): undefined reference to `getVars6FC'
collect2: error: ld returned 1 exit status
make[4]: *** [<builtin>: testupnpdescgen] Error 1
make[4]: Leaving directory '/home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510'
make[3]: *** [Makefile:75: /home/studzik/CTCGFW/openwrt/build_dir/target-mips_24kc_musl/miniupnpd-2.1.20200510/.built] Error 2
make[3]: Leaving directory '/home/studzik/CTCGFW/openwrt/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#0.38#0.12#1.62
make[2]: *** [package/Makefile:113: package/feeds/packages/miniupnpd/compile] Error 2
make[2]: Leaving directory '/home/studzik/CTCGFW/openwrt'
make[1]: *** [package/Makefile:107: /home/studzik/CTCGFW/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/studzik/CTCGFW/openwrt'
make: *** [/home/studzik/CTCGFW/openwrt/include/toplevel.mk:227: world] Błąd 2
`

AmadeusGhost, 2020-09-11:
In fact, openwrt does not support setting the "Enable IPv6 support in packages" config to n, which will break the compilation of many packages.

1715173329, 2020-09-14:
I will improve the detection on this weekend if free.

closed 2020-09-19 by 1715173329