immortalwrt/immortalwrt #205: Htop依赖问题

ksong008, 2020-09-14:


AmadeusGhost, 2020-09-14:
哪个分支？

ksong008, 2020-09-14:
@AmadeusGhost master，k5.4都一样

AmadeusGhost, 2020-09-14:
[image]
我在 k5.4 （ipq40xx）上试了下好像没问题诶

ksong008, 2020-09-14:
@AmadeusGhost 5.4是别人反馈的，今天我试了下master，boom
ps:能不能回群里说，这比tg还tg

AmadeusGhost, 2020-09-14:
u1s1，issue 反馈 bug 更为合适，加上最近事情太多了（天天都不在宿舍那种），我也只能只看 issue
跑了下应该并不是 htop 的问题，master 上 samba 36 是炸了，我等会看下 commit，直接 drop 掉算了

ksong008, 2020-09-14:
你 make clean 跑一下看看

ksong008, 2020-09-14:
> u1s1，issue 反馈 bug 更为合适，加上最近事情太多了（天天都不在宿舍那种），我也只能只看 issue
> 跑了下应该并不是 htop 的问题，master 上 samba 36 是炸了，我等会看下 commit，直接 drop 掉算了
你抽空，make clean跑下看看

AmadeusGhost, 2020-09-14:
> 你抽空，make clean跑下看看

就是全新编译的 master 代码😂

AmadeusGhost, 2020-09-14:
> 你抽空，make clean跑下看看

已测试移除 samba36 后 ipq40xx 全新编译正常

1715173329, 2020-09-14:
> master 上 samba 36 是炸了

@AmadeusGhost 我记得这个不是早就被 drop 掉了么。。

AmadeusGhost, 2020-09-14:
> 我记得这个不是早就被 drop 掉了么。。

官方是早 drop 了，不过之前 merge 的时候跳过了，毕竟 samba4 编译要多不少时间（小声
[image]

ksong008, 2020-09-14:
> [image]
> 我在 k5.4 （ipq40xx）上试了下好像没问题诶
[image]

1715173329, 2020-09-14:
> > 我记得这个不是早就被 drop 掉了么。。
> 
> 官方是早 drop 了，不过之前 merge 的时候跳过了，毕竟 samba4 编译要多不少时间（小声
> [image]

天天刷这张烂图有意思¿

1715173329, 2020-09-14:
另外麻烦以后 merge 不要混杂其他无关更改，否则将会难以定位

AmadeusGhost, 2020-09-14:
> [image]

这个我 master 编译是能过的😂

> 天天刷这张烂图有意思¿

您觉得是烂图那把我 t 了即可，眼不见心为静

MYMLD, 2020-09-14:
你们别吵了，图片是我的，确定如此，UB系统20. dl是全的，真过不了，已经删除了整个HTOP文件夹，然后编译 一切正常，
没更新之前上个分支K5.4 内核是5.461一切正常 ，后来更新到现在的版本5.465 分支K5.4 及主分支 同样的报错～K54分支删除HTOP目录编译一切正常，现在编译的是主分支，同样必须删除了HTOP才能过，先删除的是HTOP目录中的001xxx文件，过不了，只有整个目录删除了 ，偶是小白````勿取笑拉

AmadeusGhost, 2020-09-14:
20.04 啊，那确实不大一样

1715173329, 2020-09-14:
@MYMLD 您的图片无问题

htop 来自于上游源码，目前我在 rockchip (based on Ubuntu 18.04) 上测试正常，其他的未详细测试
这周有空我看看什么情况，或者等 @AmadeusGhost 大佬看看。

1715173329, 2020-09-14:
> > 天天刷这张烂图有意思¿
> 
> 您觉得是烂图那把我 t 了即可，眼不见心为静

你大可

1715173329, 2020-09-14:
> > 天天刷这张烂图有意思¿
> 
> 您觉得是烂图那把我 t 了即可，眼不见心为静

你大可不必找任何借口或理由搪塞或转移话题，我只想讨论问题本身

ksong008, 2020-09-14:
> 20.04 啊，那确实不大一样

其实，我早上也失败的，删了patch之后通过，然后没法复原了…哪怕是make clean...

AmadeusGhost, 2020-09-14:
> 其实，我早上也失败的，删了patch之后通过，然后没法复原了…哪怕是make clean...

说真的，信息最好能给多少给多少，不然不知道具体啥问题 😂
> 你大可

好啊，反正咱对于这里而言只有 noise，这边建议您直接 reply to me with /ban in every groups.

1715173329, 2020-09-14:
> 好啊，反正咱对于这里而言只有 noise，这边建议您直接 reply to me with /ban in every groups.

继续阴阳怪气 & 转移话题，请

AmadeusGhost, 2020-09-14:
> 继续阴阳怪气 & 转移话题，请

？您爱咋想就咋想，不过吧，咱只希望 “我可以送你出去” 那个“你”指的是我而不是 reply to others.

1715173329, 2020-09-15:
> > 继续阴阳怪气 & 转移话题，请
> 
> ？您爱咋想就咋想，不过吧，咱只希望 “我可以送你出去” 那个“你”指的是我而不是 reply to others.

你也大可不必动不动就拿退出来威胁谁
也不必装作弱势像是我 qj 了你一样
也不必说得好像我没事喜欢迫害别人一样

如果你自己都不认为自己是 member，我再挽留也毫无意义

AmadeusGhost, 2020-09-15:
> 你也大可不必动不动就拿退出来威胁谁

威胁…？我求您把我拉黑了行不行

closed 2020-09-15 by AmadeusGhost