immortalwrt/immortalwrt #207: master 分支github Actions云编译ipq40xx p2w_r619ac-128m 错误

PassWall-OpenWrt, 2020-09-26:
2020-09-26T10:50:25.5603182Z Applying /workdir/openwrt/target/linux/ipq40xx/patches-5.4/901-arm-boot-add-dts-files.patch using plaintext: 
2020-09-26T10:50:25.5613263Z patch: **** malformed patch at line 68:  	qcom-msm8660-surf.dtb \
2020-09-26T10:50:25.5613872Z patching file arch/arm/boot/dts/Makefile
2020-09-26T10:50:25.5614346Z 
2020-09-26T10:50:25.5616952Z Patch failed!  Please fix /workdir/openwrt/target/linux/ipq40xx/patches-5.4/901-arm-boot-add-dts-files.patch!
2020-09-26T10:50:25.5619423Z Makefile:25: recipe for target '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.67/.prepared_495d2177806cd6589ab282d7a843cdf3' failed
2020-09-26T10:50:25.5621026Z make[4]: Leaving directory '/workdir/openwrt/target/linux/ipq40xx'
2020-09-26T10:50:25.5622005Z Makefile:13: recipe for target 'compile' failed
2020-09-26T10:50:25.5623409Z make[4]: *** [/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.67/.prepared_495d2177806cd6589ab282d7a843cdf3] Error 1
2020-09-26T10:50:25.5624431Z make[3]: *** [compile] Error 2
2020-09-26T10:50:25.5631801Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2020-09-26T10:50:25.5655769Z time: target/linux/compile#10.43#4.77#15.95
2020-09-26T10:50:25.5656403Z make[2]: *** [target/linux/compile] Error 2
2020-09-26T10:50:25.5658309Z make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
2020-09-26T10:50:25.5659248Z target/Makefile:23: recipe for target 'target/linux/compile' failed
2020-09-26T10:50:25.5660145Z make[2]: Leaving directory '/workdir/openwrt'
2020-09-26T10:50:25.5661322Z target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile' failed
2020-09-26T10:50:25.5679201Z make[1]: Leaving directory '/workdir/openwrt'
2020-09-26T10:50:25.5682622Z make: *** [world] Error 2
2020-09-26T10:50:25.5683409Z /workdir/openwrt/include/toplevel.mk:234: recipe for target 'world' failed
2020-09-26T10:50:25.5689514Z ##[error]Process completed with exit code 2.

完整的log在 [https://paste.ubuntu.com/p/MD5m9YsGYh/](url)

ksong008, 2020-09-26:
请把补丁文件@@ -837,11 +837,54 @@中的54改成56试试

AmadeusGhost, 2020-09-26:
这次 git merge 的时候这里没冲突就没注意到了 Orz

commit fac29ebd (2020-09-26): ipq40xx: refresh patches

fix #207
closed 2020-09-26 by AmadeusGhost