immortalwrt/immortalwrt #214: 筋斗云编译不过了

outemancut, 2020-10-21:
MODPOST vmlinux.o
  KSYM    .tmp_kallsyms1.o
  KSYM    .tmp_kallsyms2.o
  LD      vmlinux
  SORTEX  vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm/boot/Image
  Kernel: arch/arm/boot/Image is ready
  XZKERN  arch/arm/boot/compressed/piggy_data
  AS      arch/arm/boot/compressed/piggy.o
  LD      arch/arm/boot/compressed/vmlinux
  OBJCOPY arch/arm/boot/zImage
  Kernel: arch/arm/boot/zImage is ready
make[5]: Nothing to be done for 'Image'.
make[6]: *** No rule to make target 'arch/arm/boot/dts/qcom-ipq4019-ap.dk04.1-c1.dtb', needed by '__build'.  Stop.
arch/arm/Makefile:351: recipe for target 'dtbs' failed
make[5]: *** [dtbs] Error 2
make[5]: Leaving directory '/root/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.202'
Makefile:24: recipe for target '/root/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.202/.image' failed
make[4]: *** [/root/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.202/.image] Error 2
make[4]: Leaving directory '/root/openwrt/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/root/openwrt/target/linux'
time: target/linux/install#10.22#6.36#17.93
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/root/openwrt'
target/Makefile:19: recipe for target '/root/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/root/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:234: recipe for target 'world' failed
make: *** [world] Error 2

1715173329, 2020-10-21:
@outemancut 什么分支？

outemancut, 2020-10-21:
openwrt-18.06-k5.4

outemancut, 2020-10-21:
> @outemancut 什么分支？

openwrt-18.06-k5.4

1715173329, 2020-10-21:
@outemancut 5.4 用 4.14 内核¿

commit d7c06b90 (2020-10-21): ipq40xx: drop kernel 4.14 support

kernel 4.14 is no longer supported with existing DTS files.

Fixes: #214
closed 2020-10-21 by 1715173329