immortalwrt/immortalwrt #216: 请问这个是什么错误啊？同样配置早上编译没问题，下午开始就一直编译不成功了

281677160, 2020-10-25:
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip" "6d141b84a541e31ed7d3cd4d938987fdfa7988d5081d3089bc1509a43d6b742f" "gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip" "https://chrome-infra-packages.appspot.com/dl/gn/gn/linux-amd64/+"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://chrome-infra-packages.appspot.com/dl/gn/gn/linux-amd64/+/gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 400 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/gn-git_revision\:e327ffdc503815916db2543ec000226a8df45163.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [/workdir/openwrt/dl/gn-git_revision:e327ffdc503815916db2543ec000226a8df45163.zip] Error 2
Makefile:131: recipe for target '/workdir/openwrt/dl/gn-git_revision:e327ffdc503815916db2543ec000226a8df45163.zip' failed
make[3]: Leaving directory '/workdir/openwrt/package/ctcgfw/naiveproxy'
time: package/ctcgfw/naiveproxy/compile#0.25#0.08#1.25
make[2]: *** [package/ctcgfw/naiveproxy/compile] Error 2
package/Makefile:111: recipe for target 'package/ctcgfw/naiveproxy/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:234: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

ksong008, 2020-10-25:
@281677160 你梯子问题

1715173329, 2020-10-25:
Makefile 编写出了点问题，将于最近修复

281677160, 2020-10-25:
> Makefile 编写出了点问题，将于最近修复

谢谢大神回答

blueveryday, 2020-10-25:
需要编译的软件包没有下载完整就会出现这个情况。
terminal走的网络不会从电脑的全局代理服务器走下载，（假设你已经开了Qv2ray这类软件）

解决方法：
启动电脑的代理服务器，比如Qv2ray这类软件。
安装如下工具：
sudo apt-get install proxychains

进入/etc下面，修改proxychains的配置文件config。
有两个文件，一个是proxychains，一个是proxychains4的配置文件，我都改成本地电脑Qv2ray的socks5的127.0.0.1:1089这个地址。

然后在terminal界面，执行命令：
proxychains4 make -j1 V=s
就行了。
这时候terminal就强制走127.0.0.1:1089这个socks5本地代理地址联网。
等走完编译流程，这时候在编译naiveproxy的时候，所需要下载的软件包就能够成功走本地代理地址下载所需要的软件包了。
只要下载文件不失败，不出错，就可以顺利编译好固件了。

基本上很多失败都是因为所需要的软件包没有下载成功引起的，亲测有效。

281677160, 2020-10-25:
> Makefile 编写出了点问题，将于最近修复

谢谢您刚才的回复，还有一个问题，就是lienol文件夹的luci-app-webrestriction跟lean文件夹的luci-app-accesscontrol功能好像是一样的，都是上网时间控制

281677160, 2020-10-25:
> 需要编译的软件包没有下载完整就会出现这个情况。
> terminal走的网络不会从电脑的全局代理服务器走下载，（假设你已经开了Qv2ray这类软件）
>
> 解决方法：
> 启动电脑的代理服务器，比如Qv2ray这类软件。
> 安装如下工具：
> sudo apt-get install proxychains
>
> 进入/etc下面，修改proxychains的配置文件config。 [...]

谢谢大神，我用的云编译，本地编译搞不过来，用的还是20年前的单核电脑

closed 2020-10-25 by 1715173329

1715173329, 2020-10-25:
@281677160 fixed

commit 89941c8f (2020-10-25): naiveproxy: fix var setting

Fixes: #216.
commit f1e925a5 (2020-10-25): naiveproxy: fix var setting

Fixes: #216.
commit 631e78e0 (2020-10-25): naiveproxy: fix var setting

Fixes: #216.