immortalwrt/immortalwrt #217: 分支18.06-k5.4编译cryptodev-linux报错

blueveryday, 2020-10-26:
[image]

如图，求解决。

AmadeusGhost, 2020-10-26:
I'll check this in my free time.

blueveryday, 2020-10-29:
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_hiwifi_hc5962=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_LDAP=y
CONFIG_LIBCURL_LDAPS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_PACKAGE_UnblockNeteaseMusic-Go=y
# CONFIG_PACKAGE_adbyby is not set
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-date_time=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_chinadns-ng=m
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_e2fsprogs=y
# CONFIG_PACKAGE_etherwake is not set
CONFIG_PACKAGE_haproxy=m
CONFIG_PACKAGE_https-dns-proxy=m
CONFIG_PACKAGE_ipt2socks=y
CONFIG_PACKAGE_kmod-crypto-acompress=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-ntfs=y
CONFIG_PACKAGE_kmod-ledtrig-default-on=y
CONFIG_PACKAGE_kmod-ledtrig-heartbeat=y
CONFIG_PACKAGE_kmod-ledtrig-netdev=y
CONFIG_PACKAGE_kmod-ledtrig-timer=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-lib-lzo=y
# CONFIG_PACKAGE_kmod-nf-conntrack-netlink is not set
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-extras=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-zram=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libblkid=y
CONFIG_PACKAGE_libcares=m
CONFIG_PACKAGE_libcomerr=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libext2fs=y
CONFIG_PACKAGE_libltdl=m
CONFIG_PACKAGE_liblua5.3=m
CONFIG_PACKAGE_libmaxminddb=m
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libnss=y
CONFIG_PACKAGE_libopenldap=y
CONFIG_PACKAGE_libsasl2=y
CONFIG_PACKAGE_libsqlite3=y
CONFIG_PACKAGE_libss=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_lua-maxminddb=m
CONFIG_PACKAGE_luasocket=m
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-adbyby-plus is not set
# CONFIG_PACKAGE_luci-app-adguardhome_INCLUDE_binary is not set
# CONFIG_PACKAGE_luci-app-arpbind is not set
# CONFIG_PACKAGE_luci-app-autoreboot is not set
# CONFIG_PACKAGE_luci-app-ddns is not set
# CONFIG_PACKAGE_luci-app-filetransfer is not set
CONFIG_PACKAGE_luci-app-naiveproxy=m
CONFIG_PACKAGE_luci-app-ngrokc=y
# CONFIG_PACKAGE_luci-app-nlbwmon is not set
CONFIG_PACKAGE_luci-app-nps=y
CONFIG_PACKAGE_luci-app-p910nd=y
CONFIG_PACKAGE_luci-app-passwall=m
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ChinaDNS_NG=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_GO=y
CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray=y
# CONFIG_PACKAGE_luci-app-ramfree is not set
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-udpxy=y
# CONFIG_PACKAGE_luci-app-unblockmusic is not set
CONFIG_PACKAGE_luci-app-unblockneteasemusic-go=y
# CONFIG_PACKAGE_luci-app-upnp is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
CONFIG_PACKAGE_luci-app-vssr=m
CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Trojan=y
# CONFIG_PACKAGE_luci-app-wol is not set
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-i18n-naiveproxy-zh-cn=m
CONFIG_PACKAGE_luci-i18n-ngrokc-zh-cn=y
CONFIG_PACKAGE_luci-i18n-nps-zh-cn=y
CONFIG_PACKAGE_luci-i18n-p910nd-zh-cn=y
# CONFIG_PACKAGE_luci-lib-fs is not set
CONFIG_PACKAGE_luci-proto-ipv6=y
# CONFIG_PACKAGE_miniupnpd is not set
CONFIG_PACKAGE_naiveproxy=y
CONFIG_PACKAGE_ngrokc=y
# CONFIG_PACKAGE_nlbwmon is not set
CONFIG_PACKAGE_npc=y
CONFIG_PACKAGE_nspr=y
CONFIG_PACKAGE_p910nd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_shadowsocks-libev-ss-server=m
CONFIG_PACKAGE_shadowsocksr-libev-server=m
CONFIG_PACKAGE_ssocks=m
CONFIG_PACKAGE_trojan=y
CONFIG_PACKAGE_trojan-go=m
CONFIG_PACKAGE_trojan-plus=m
CONFIG_PACKAGE_udpxy=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2ray=m
# CONFIG_PACKAGE_vsftpd-alt is not set
# CONFIG_PACKAGE_wol is not set
CONFIG_PACKAGE_zram-swap=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_JSON1=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TROJAN_GO_COMPRESS_UPX=y
CONFIG_V2RAY_COMPRESS_UPX=y
CONFIG_V2RAY_EXCLUDE_ASSETS=y
CONFIG_V2RAY_EXCLUDE_V2CTL=y
CONFIG_V2RAY_JSON_INTERNAL=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y

1715173329, 2020-10-30:
Everything is functional.

See https://github.com/project-openwrt/OpenWrt-Compilation-Checker/runs/1326861117?check_suite_focus=true

AmadeusGhost, 2020-11-04:
@blueveryday 此分支上的 cryptodev-linux 没有编译问题。我复制了您的 .config 看了下，恕我直言，可能是跟您重量级的插件选择太多了有关系，毕竟这是 mips 而不是 x86 (((

closed 2020-11-04 by AmadeusGhost

blueveryday, 2020-11-05:
> Everything is functional.
> 
> See https://github.com/project-openwrt/OpenWrt-Compilation-Checker/runs/1326861117?check_suite_focus=true

谢谢。