immortalwrt/immortalwrt #218: Nanopi neo plus2 USB问题 缺少kmod-usb-lib-composite

CMCCAdmin, 2020-10-29:
我的设备时nanopi neo plus2 全志h5芯片，编译后有一个usb口插上设备没有任何反应，dmesg没有任何输出。看资料说这个口是otg专用接口，但前几天我用Lienol的5.4源不知怎么修改的配置usb突然能用了，找了好久发现可能是以下这些包起了作用。
CONFIG_PACKAGE_kmod-usb-gadget=y
CONFIG_PACKAGE_kmod-usb-gadget-cdc-composite=y
CONFIG_PACKAGE_kmod-usb-gadget-eth=y
CONFIG_PACKAGE_kmod-usb-gadget-hid=y
CONFIG_PACKAGE_kmod-usb-gadget-mass-storage=y
CONFIG_PACKAGE_kmod-usb-gadget-serial=y
在你的源和lean的源中make menuconfig 看不到这些包，所以只能改.config配置文件，然而缺少依赖libcomposite.ko，kmod-usb-lib-composite  把这个依赖加在配置文件中没用，应该是源码中没有这个包，不知道改怎么解决

Package kmod-usb-gadget-eth is missing dependencies for the following libraries:
libcomposite.ko

make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/ubuntu/Desktop/openwrt/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.56#0.19#0.72
make[3]: Entering directory '/home/ubuntu/Desktop/openwrt/package/kernel/linux'
mkdir -p /home/ubuntu/Desktop/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-sunxi/stamp

SHELL= flock /home/ubuntu/Desktop/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/ubuntu/Desktop/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/packages/.pkgdir/kernel/. /home/ubuntu/Desktop/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-sunxi/'
touch /home/ubuntu/Desktop/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-sunxi/stamp/.kernel_installed
Package kmod-usb-gadget-eth is missing dependencies for the following libraries:
libcomposite.ko
make[3]: *** [modules/usb.mk:180: /home/ubuntu/Desktop/openwrt/bin/targets/sunxi/cortexa53/packages/kmod-usb-gadget-eth_5.4.72-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/ubuntu/Desktop/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#4.40#0.62#4.98
make[2]: *** [package/Makefile:114: package/kernel/linux/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/Desktop/openwrt'
make[1]: *** [package/Makefile:108: /home/ubuntu/Desktop/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/Desktop/openwrt'
make: *** [/home/ubuntu/Desktop/openwrt/include/toplevel.mk:236: world] Error 2

CMCCAdmin, 2020-10-29:
配置文件添加这个依赖后提示这个，之后报错
WARNING: can't parse line: CONGIF_PACKAGE_kmod-usb-lib-composite=y

1715173329, 2020-10-30:
所在分支？

CMCCAdmin, 2020-10-30:
> 所在分支？

master分支

AmadeusGhost, 2020-10-31:
This target can't enable the corresponding config.

closed 2020-10-31 by AmadeusGhost