immortalwrt/immortalwrt #220: 源码某插件好像有问题

281677160, 2020-11-02:
同时编译5个不一样的机型openwrt，有2个能编译成功，有3个编译不成功，.config互换机型编译，本来成功的失败了，失败的成功编译，也就是选择了某插件导致不能编译成功的，openwrt-18.06跟openwrt-18.06-k5.4分支都同样问题，请大神看看什么问题

Configuring luci-i18n-nlbwmon-zh-cn.
make[2]: *** [package/install] Error 255
package/Makefile:66: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
package/Makefile:109: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:234: recipe for target 'world' failed

outemancut, 2020-11-02:
我只知道石像鬼QOS不能正常使用了LUCI不能正常调用 
5.4内核的之前提示缺少'kmod-sched-cake-virtual'
我现在用18.06这个分支了不用5.4了 还是18.06这个分支比较稳定一点
你这种问题一般都是编译的缓存文件错误 你把报错的编译过的文件删了再编译试试

281677160, 2020-11-02:
同样配置，前天还是大前天还可以编译成功的，昨天跟今天就不行

AmadeusGhost, 2020-11-03:
某插件是啥插件啊 ...？就不能 `make V=s -j1` 指出具体哪里报错吗？

closed 2020-11-04 by 1715173329