immortalwrt/immortalwrt #223: 筋斗云固件18.06分支有问题

outemancut, 2020-11-04:
酸酸乳 Plus+ 无法启动 4.14内核

outemancut, 2020-11-04:
Wed Nov  4 20:09:42 2020 daemon.err uhttpd[1623]: sed: /tmp/dnsmasq.ssr/gfw_base.conf: No such file or directory
Wed Nov  4 20:09:42 2020 daemon.err uhttpd[1623]: /usr/share/shadowsocksr/gfw2ipset.sh: line 38: can't create /tmp/dnsmasq.ssr/blacklist_forward.conf: nonexistent directory
Wed Nov  4 20:09:42 2020 daemon.err uhttpd[1623]: /usr/share/shadowsocksr/gfw2ipset.sh: line 39: can't create /tmp/dnsmasq.ssr/blacklist_forward.conf: nonexistent directory
Wed Nov  4 20:09:42 2020 daemon.err uhttpd[1623]: /usr/share/shadowsocksr/gfw2ipset.sh: line 40: can't create /tmp/dnsmasq.ssr/whitelist_forward.conf: nonexistent directory
Wed Nov  4 20:09:42 2020 daemon.err uhttpd[1623]: /usr/share/shadowsocksr/gfw2ipset.sh: line 41: can't create /tmp/dnsmasq.ssr/denylist.conf: nonexistent directory

AmadeusGhost, 2020-11-04:
这跟竞斗云和 4.14 内核都没关系吧 ,,, 个人希望没事不要把插件的问题指向内核或者机器。

8688Add, 2020-11-04:
> 这跟竞斗云和 4.14 内核都没关系吧 ,,, 个人希望没事不要把插件的问题指向内核或者机器。

今天编译了个vssr测试了一下18.06分支,4.14内核这几天的更新确实无法启动vssr

zxlhhyccc, 2020-11-04:
首先你要说清楚是哪个分支编译的？如果是1907或者master分支，要修改init的一个代码。。。。

outemancut, 2020-11-04:
> 这跟竞斗云和 4.14 内核都没关系吧 ,,, 个人希望没事不要把插件的问题指向内核或者机器。

难道反馈问题连机器内核都不需要说明一下？之前18.06 4.14内核的高通4019的筋斗云可以使用 这两天更新的不能使用了 反馈一下说是指内核机器问题？

zxlhhyccc, 2020-11-04:
请将init的这行代码改成
[image]

outemancut, 2020-11-04:
> 请将init的这行代码改成
> [image]

OK 我试试

AmadeusGhost, 2020-11-04:
> 反馈一下说是指内核机器问题？

是我太菜了，理解能力堪忧，不过您原来的标题是得改改，我还以为是内核炸了
> 首先你要说清楚是哪个分支编译的？
他们不都说了是 18.06 分支嘛

zxlhhyccc, 2020-11-04:
> > 请将init的这行代码改成
> > [image]
> 
> OK 我试试

应该可以了吧。。。。

outemancut, 2020-11-04:
find bin

> > > 请将init的这行代码改成
> > > [image]
> > 
> > 
> > 好吧我试试
> 
> 应该可以了吧....

我又重新编译了一下 刷上还是一样无法启动
[image]

1715173329, 2020-11-04:
@AmadeusGhost

最近确实又有一堆人因为“内核升级”而出现各种问题了。。
可能是最近某个 commit 导致的吧 (((

zxlhhyccc, 2020-11-04:
> @AmadeusGhost
> 
> 最近确实又有一堆人因为“内核升级”而出现各种问题了。。
> 可能是最近某个 commit 导致的吧 (((

我怀疑是那个include/netfilter.mk问题，config-5.4已经屏蔽了CONFIG_IP6_NF_TARGET_MASQUERADE，我改回去试试。。。。

AmadeusGhost, 2020-11-04:
> 我怀疑是那个include/netfilter.mk问题，config-5.4已经屏蔽了CONFIG_IP6_NF_TARGET_MASQUERADE，我改回去试试。。。。

您到底在干啥啊？上面说的是 18.06 您往 5.4 上扯。别改回去试了，您先看看官方的 commit message 再说，还可以去搜 linux kernel 的 commit message，再不行 make kernel_menuconfig 找 symbol，总比 xj8 猜好得多。

zxlhhyccc, 2020-11-04:
> > 我怀疑是那个include/netfilter.mk问题，config-5.4已经屏蔽了CONFIG_IP6_NF_TARGET_MASQUERADE，我改回去试试。。。。
> 
> 您到底在干啥啊？上面说的是 18.06 您往 5.4 上扯。别改回去试了，您先看看官方的 commit message 再说，还可以去搜 linux kernel 的 commit message，再不行 make kernel_menuconfig 找 symbol，总比 xj8 猜好得多。

因为我前天master的73内核正常。。。。

xiaobaishu1, 2020-11-04:
> @AmadeusGhost
> 
> 最近确实又有一堆人因为“内核升级”而出现各种问题了。。
> 可能是最近某个 commit 导致的吧 (((

一号那天用的 最新源码没问题 我就两天没编译而已  不过我发现了节点那里点更新节点后 第一个节点自动应用在主服务器上面 而日志上面没有显示完全 2020-10-31 10:37:09 成功解析: v2ray 节点, 沪港CN2隧道|香港电讯
2020-10-31 10:37:03 服务正在暂停
之前解析完成会有这一项
2020-10-31 10:37:03 服务正在暂停
------------end------------
------------end------------
-----------start-----------
-----------start-----------

zxlhhyccc, 2020-11-04:
> 一号那天用的 最新源码没问题 我就两天没编译而已 不过我发现了节点那里点更新节点后 第一个节点自动应用在主服务器上面 而日志上面没有显示完全 2020-10-31 10:37:09 成功解析: v2ray 节点, 沪港CN2隧道|香港电讯
> 2020-10-31 10:37:03 服务正在暂停
> 之前解析完成会有这一项
> 2020-10-31 10:37:03 服务正在暂停 [...]

1号的肯定是没问题的

1715173329, 2020-11-05:
问题源于上游 commit 1a69f50dc627f6fc5860f871edec357cc0a187e6。
init 中对于附加命令的定义发生了改动，故导致部分 init 失效

zxlhhyccc, 2020-11-05:
> 问题源于上游 commit [1a69f50](https://github.com/project-openwrt/openwrt/commit/1a69f50dc627f6fc5860f871edec357cc0a187e6)。
> init 中对于附加命令的定义发生了改动，故导致部分 init 失效

将init中得代码：EXTRA_COMMANDS=rules 改成： EXTRA_COMMAND "rules"即可解决问题。。。

closed 2020-11-11 by 1715173329