immortalwrt/immortalwrt #236: Passwall 和 SSR 出错

lgfyxx1, 2020-12-09:
出错代码：

* check_data_file_clashes: Package xray wants to install file /workdir/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/v2ray
	But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_data_file_clashes: Package xray wants to install file /workdir/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/v2ray
	But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package xray.
 * check_data_file_clashes: Package xray wants to install file /workdir/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/bin/v2ray
	But that file is already provided by package  * v2ray
 * opkg_install_cmd: Cannot install package luci-app-passwall.

默认官方编译的，什么也没变动，是V2ray和Xray不能同时选嘛？

1715173329, 2020-12-09:
二选一，或者取消 xray 的兼容模式

closed 2020-12-09 by 1715173329

lgfyxx1, 2020-12-09:
二选一了还是不行，只安装SSR一样的错误，并且是云编译，求大神支招。