immortalwrt/immortalwrt #239: openwrt-19.07 分支 GitHub Actions 编译 x86_64 OpenClash 在 luci 界面找不到 OpenClash

PassWall-OpenWrt, 2020-12-13:
版本 r12429-e3fe7f8657

config 文件内容 https://paste.ubuntu.com/p/2mQd6RWFTC/

[image]

[image]

zxlhhyccc, 2020-12-23:
> 版本 r12429-e3fe7f8657
> 
> config 文件内容 https://paste.ubuntu.com/p/2mQd6RWFTC/
> 
> [image]
> 
> [image]

你继续使用0.41.06？

PassWall-OpenWrt, 2020-12-23:
> > 版本 r12429-e3fe7f8657
> > config 文件内容 https://paste.ubuntu.com/p/2mQd6RWFTC/
> > [image]
> > [image]
> 
> 你继续使用0.41.06？

I opened this issue 10 days ago !

PassWall-OpenWrt, 2020-12-24:
@AmadeusGhost 这个已经解决了吗？就关了？

1715173329, 2020-12-24:
This still doesn't work, so I would like to reopen it until resolved.

commit 5f0a5f74 (2020-12-25): OpenClash: bump to v0.41.13-beta

Fixes: #239
closed 2020-12-25 by 1715173329