immortalwrt/immortalwrt #242: cryptopp820编译出错

ksong008, 2020-12-21:
1806-k5.4
cryptopp820编译出错
[image]

AmadeusGhost, 2020-12-21:
Duplicate with #180

ksong008, 2020-12-21:
> Duplicate with #180

重新clone来也是报错。。。

AmadeusGhost, 2020-12-21:
试试 `git pull && git cherry-pick 162b53635815e7d606cf7627d66a25ea52d11e89`
[image]

ksong008, 2020-12-21:
> 试试 `git pull && git cherry-pick 162b53635815e7d606cf7627d66a25ea52d11e89`
> [image]

可能是config的问题，我load老的config就出错，我新建了一个pass

closed 2020-12-21 by ksong008