immortalwrt/immortalwrt #243: libcryptopp: update to 8.3.0编译报错！！！

zxlhhyccc, 2020-12-22:
[image]

hue715, 2020-12-22:
试了一下，master分支下单独编译libcryptopp 8.3.0编译没问题。

AmadeusGhost, 2020-12-23:
您好歹截图的时候往上截一点啊，我这边就看到又是这个 aria_simd.cpp.o failed

zxlhhyccc, 2020-12-23:
> 您好歹截图的时候往上截一点啊，我这边就看到又是这个 aria_simd.cpp.o failed，具体啥报错都不知道

````
In file included from /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:18:
/home/lin/openwrt-luci-master/staging_dir/toolchain-arm_cortex-a9_gcc-10.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/10.2.0/include/arm_neon.h:31:2: error: #error "NEON intrinsics not available with the soft-float ABI.  Please use -mfloat-abi=softfp or -mfloat-abi=hard"
   31 | #error "NEON intrinsics not available with the soft-float ABI.  Please use -mfloat-abi=softfp or -mfloat-abi=hard"
      |  ^~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:67:34: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   67 | inline void ARIA_GSRK_NEON(const uint32x4_t X, const uint32x4_t Y, byte RK[16])
      |                                  ^~~~~~~~~~
      |                                  uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:67:54: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   67 | inline void ARIA_GSRK_NEON(const uint32x4_t X, const uint32x4_t Y, byte RK[16])
      |                                                      ^~~~~~~~~~
      |                                                      uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp: In function 'void CryptoPP::ARIA_UncheckedSetKey_Schedule_NEON(CryptoPP::byte*, CryptoPP::word32*, unsigned int)':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:82:8: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   82 |  const uint32x4_t w0 = vld1q_u32(ws+ 0);
      |        ^~~~~~~~~~
      |        uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:83:8: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   83 |  const uint32x4_t w1 = vld1q_u32(ws+ 8);
      |        ^~~~~~~~~~
      |        uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:84:8: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   84 |  const uint32x4_t w2 = vld1q_u32(ws+12);
      |        ^~~~~~~~~~
      |        uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:85:8: error: 'uint32x4_t' does not name a type; did you mean 'uint32_t'?
   85 |  const uint32x4_t w3 = vld1q_u32(ws+16);
      |        ^~~~~~~~~~
      |        uint32_t
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:87:21: error: 'w0' was not declared in this scope; did you mean 'ws'?
   87 |  ARIA_GSRK_NEON<19>(w0, w1, rk +   0);
      |                     ^~
      |                     ws
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:87:25: error: 'w1' was not declared in this scope; did you mean 'ws'?
   87 |  ARIA_GSRK_NEON<19>(w0, w1, rk +   0);
      |                         ^~
      |                         ws
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:88:25: error: 'w2' was not declared in this scope; did you mean 'ws'?
   88 |  ARIA_GSRK_NEON<19>(w1, w2, rk +  16);
      |                         ^~
      |                         ws
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:89:25: error: 'w3' was not declared in this scope; did you mean 'ws'?
   89 |  ARIA_GSRK_NEON<19>(w2, w3, rk +  32);
      |                         ^~
      |                         ws
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp: In function 'void CryptoPP::ARIA_ProcessAndXorBlock_NEON(const byte*, CryptoPP::byte*, const byte*, CryptoPP::word32*)':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:138:5: error: 'vld1q_u8' was not declared in this scope
  138 |     vld1q_u8(xorBlock),
      |     ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:141:6: error: 'vrev32q_u8' was not declared in this scope
  141 |      vrev32q_u8(vld1q_u8((rk))))));
      |      ^~~~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:139:5: error: 'veorq_u8' was not declared in this scope
  139 |     veorq_u8(
      |     ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:137:4: error: 'veorq_u8' was not declared in this scope
  137 |    veorq_u8(
      |    ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:136:3: error: 'vst1q_u8' was not declared in this scope
  136 |   vst1q_u8(outBlock,
      |   ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/libcryptopp-8.3.0/aria_simd.cpp:147:5: error: 'vld1q_u8' was not declared in this scope
  147 |     vld1q_u8(outBlock),
      |     ^~~~~~~~
[... cut ...]
time: package/bf-package-master/lean/libcryptopp/compile#33.98#5.90#51.25
    ERROR: package/bf-package-master/lean/libcryptopp failed to build.
package/Makefile:112: recipe for target 'package/bf-package-master/lean/libcryptopp/compile' failed
make[1]: *** [package/bf-package-master/lean/libcryptopp/compile] Error 1
make[1]: Leaving directory '/home/lin/openwrt-luci-master'
/home/lin/openwrt-luci-master/include/toplevel.mk:240: recipe for target 'package/luci-app-amule/compile' failed
make: *** [package/luci-app-amule/compile] Error 2
````````

AmadeusGhost, 2020-12-23:
我发现这个包就算回退到 8.2.0 在 arm 上一样编译不过
[image]

zxlhhyccc, 2020-12-23:
> 我发现这个包就算回退到 8.2.0 在 arm 上一样编译不过
> [image]

8.2.0一直稳定编译正常。。。。

AmadeusGhost, 2020-12-24:
> 8.2.0一直稳定编译正常。。。。

什么叫正常？加个 DEPENDS:=@!HAS_FPU @arm 算不算？
再试试最新 commit，不行回退到 5.6.4 算了

zxlhhyccc, 2020-12-24:
最新 commit编译通过。。。。

closed 2020-12-24 by AmadeusGhost

1715173329, 2020-12-24:
垃圾博通不支持 neon 罢了

AmadeusGhost, 2020-12-25:
> 垃圾博通不支持 neon 罢了

是因为这个编译失败，原项目已改为使用 `CRYPTOPP_ARMV7A_NEON`，不过这个 cmake 在 armv7 上也是编译失败的，似乎无法正确链接编译器。