immortalwrt/immortalwrt #247: luci-app-zerotier eror One or more required fields have no value!

predators46, 2021-01-02:
Plis fix luci zero tier openwrt branch 19.07.2

1715173329, 2021-01-02:
> Plis fix luci zero tier openwrt branch 19.07.2

Any screenshot?

predators46, 2021-01-02:
> > Plis fix luci zero tier openwrt branch 19.07.2
> 
> Any screenshot?

[image]

1715173329, 2021-01-02:
Click '+' besides the inputbox and then apply the modification.

closed 2021-01-02 by 1715173329

predators46, 2021-01-02:
> Click '+' besides the inputbox and then apply the modification.

thanks friends solved. there is more you have to give permission in the file /etc/init.d/zerotier

1715173329, 2021-01-02:
> there is more you have to give permission in the file /etc/init.d/zerotier

It should be fixed in latest git HEAD, so just please update your feeds.

predators46, 2021-01-02:
> > there is more you have to give permission in the file /etc/init.d/zerotier
> 
> It should be fixed in latest git HEAD, so just please update your feeds.

Thanks my friend. can you add luci to tun2socks my client needs this

1715173329, 2021-01-02:
> Thanks my friend. can you add luci to tun2socks my client needs this

I'll be glad if you can submit a PR.

predators46, 2021-01-02:
> > Thanks my friend. can you add luci to tun2socks my client needs this
> 
> I'll be glad if you can submit a PR.

I don't know how to create a file. i plan to combine shadowsocks and tun2socks