immortalwrt/immortalwrt #249: 引入rtl88x2bu驱动，Lean code base，编译出错，详见后附log

esirplayground, 2021-01-05:
make[3]: Entering directory '/home/esir/lede/package/esir/rtl88x2bu-ctcgfw'
touch /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.prepared_8bba0a535dce0c1977d367e4a262ffb9_6664517399ebbbc92a37c5bb081b5c53_check
. /home/esir/lede/include/shell.sh; xzcat /home/esir/lede/dl/rtl88x2bu-2020-07-21-a2abcaf5.tar.xz | tar -C /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5

Applying ./patches/001-use-kernel-byteorder.patch using plaintext:
patching file include/drv_types.h

Applying ./patches/010-fix-build-on-powerpc.patch using plaintext:
patching file include/byteorder/swab.h

Applying ./patches/020-remove-repeat-flies.patch using plaintext:
patching file include/drv_conf.h
patching file include/linux/old_unused_rtl_wireless.h (renamed from include/linux/wireless.h)
patching file include/rtl_autoconf.h (renamed from include/autoconf.h)
touch /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.prepared_8bba0a535dce0c1977d367e4a262ffb9_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.configured_*
rm -f /home/esir/lede/staging_dir/target-aarch64_generic_musl/stamp/.rtl88x2bu-ctcgfw_installed
(cd /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/./; if [ -x ./configure ]; then find /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/ -name config.guess | xargs -r chmod u+w; find /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/ -name config.guess | xargs -r -n1 cp --remove-destination /home/esir/lede/scripts/config.guess; find /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/ -name config.sub | xargs -r chmod u+w; find /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/ -name config.sub | xargs -r -n1 cp --remove-destination /home/esir/lede/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5=rtl88x2bu-2020-07-21-a2abcaf5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5=rtl88x2bu-2020-07-21-a2abcaf5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -DRTW_SINGLE_WIPHY -DCONFIG_CONCURRENT_MODE -DCONFIG_WIFI_MONITOR" CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5=rtl88x2bu-2020-07-21-a2abcaf5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -DRTW_SINGLE_WIPHY -DCONFIG_CONCURRENT_MODE -DCONFIG_WIFI_MONITOR" CPPFLAGS="-I/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/include -I/home/esir/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/home/esir/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/home/esir/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include " LDFLAGS="-L/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/lib -L/home/esir/lede/staging_dir/target-aarch64_generic_musl/lib -L/home/esir/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/esir/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.built
touch /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/.built_check
cat /dev/null > /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/rtl88x2bu.symvers; for subdir in .; do cat /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/*.symvers 2>/dev/null > /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/$subdir/Module.symvers; done
make  -C "/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.85" KCFLAGS="-ffile-prefix-map=/home/esir/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/esir/lede/staging_dir/host/include -I/home/esir/lede/staging_dir/hostpkg/include -I/home/esir/lede/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Dec 29 14:28:04 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/esir/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/esir/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.85 M="/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5" NOSTDINC_FLAGS="-I/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5 -I/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/include -I/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport -I/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport/uapi -I/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/include/mac80211 -I/home/esir/lede/staging_dir/target-aarch64_generic_musl/usr/include/mac80211/uapi -include backport/autoconf.h -include backport/backport.h" USER_EXTRA_CFLAGS="-DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -DRTW_SINGLE_WIPHY -DCONFIG_CONCURRENT_MODE -DCONFIG_WIFI_MONITOR" CONFIG_RTL8822BU=m USER_MODULE_NAME=rtl88x2bu modules
make[4]: Entering directory '/home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.85'
  CC [M]  /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/core/rtw_cmd.o
  CC [M]  /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/core/rtw_security.o
  CC [M]  /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/core/rtw_debug.o
  CC [M]  /home/esir/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2020-07-21-a2abcaf5/core/rtw_io.o
[... cut ...]
package/Makefile:111: recipe for target 'package/esir/rtl88x2bu-ctcgfw/compile' failed
make[2]: *** [package/esir/rtl88x2bu-ctcgfw/compile] Error 2
make[2]: Leaving directory '/home/esir/lede'
package/Makefile:107: recipe for target '/home/esir/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/esir/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/esir/lede'
/home/esir/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

1715173329, 2021-01-05:
A runtime test now is launched there:
https://github.com/project-openwrt/OpenWrt-Compilation-Checker/actions/runs/462663824

I'm not sure what's the difference with the source code from each repo, but it's more reasonable to use ours to compile it and check the status.
If after that the issue still remains, please upload your configuration file here via [Ubuntu Pastebin](paste.ubuntu.com).

ElonH, 2021-01-05:
module `rtl88x2b` is compiled for x86 arch currently. I'm not guarantee it can be compiled in ARMv8

AmadeusGhost, 2021-01-05:
Lean 源码 mac80211 版本为 5.8，对应这边是 master 分支，请用 https://github.com/project-openwrt/openwrt/tree/master/package/ctcgfw/rtl88x2bu 这里的代码

1715173329, 2021-01-05:
https://github.com/project-openwrt/OpenWrt-Compilation-Checker/runs/1649211188?check_suite_focus=true#step:6:207

It works fine, so plz check twice.
I'll close this for now, and you can reopen it anytime if you have any further issues.

closed 2021-01-05 by 1715173329