immortalwrt/immortalwrt #250: luci-app-ssrserver-python can not start when it uses python3

Quzard, 2021-01-06:
luci-app-ssrserver-python can not start when it uses python3 but can start when python

1715173329, 2021-01-06:
> luci-app-ssrserver-python can not start when it uses python3 but can start when python
Could you provide any log?

Quzard, 2021-01-06:
I don't know why it happens.
When I use this project's luci-app-ssrserver-python, it will get stuck in “ /etc/init.d/firewall restart"
However when I change all codes from python3 to python, it won't happen.

I compare

1715173329, 2021-01-06:
Okay, I will have a look this weekend.
The workaround is using ssr-libev-server instead.

1715173329, 2021-01-08:
> When I use this project's luci-app-ssrserver-python, it will get stuck in “ /etc/init.d/firewall restart"

Maybe it's caused by other apps (like zerotier fixed recently), however I cannot reproduce it.
btw, which branch do you use?

1715173329, 2021-01-08:
and, there's a bug that luci status still shows "Stoppted" even if it is already running.

Quzard, 2021-01-08:
> and, there's a bug that luci status still shows "Stoppted" even if it is already running.
I use 18.06.
I will take a look at it. Thanks.

Quzard, 2021-01-08:
> and, there's a bug that luci status still shows "Stoppted" even if it is already running.

I use 18.06.
I will take a look at it. Thank you.

commit 9355dd45 (2021-01-08): luci-app-ssrserver-python: improve status check

Running `ps` via luci.exec directly will cause no output
to compare as `ps` cannot detect the size of screen, adding
`-w` to solve this.

Fixes: 0e285c3037c3481d0c1b7a7c672a6ca5692ef35f ("add lean's package")
Fixes: #250
commit e76da0eb (2021-01-08): luci-app-ssrserver-python: improve status check

Running `ps` via luci.exec directly will cause no output
to compare as `ps` cannot detect the size of screen, adding
`-w` to solve this.

Fixes: 0e285c3037c3481d0c1b7a7c672a6ca5692ef35f ("add lean's package")
Fixes: #250
commit 070b8b64 (2021-01-08): luci-app-ssrserver-python: improve status check

Running `ps` via luci.exec directly will cause no output
to compare as `ps` cannot detect the size of screen, adding
`-w` to solve this.

Fixes: 0e285c3037c3481d0c1b7a7c672a6ca5692ef35f ("add lean's package")
Fixes: #250
closed 2021-01-08 by 1715173329