immortalwrt/immortalwrt #251: [autocore] let `passwd` using `busybox` to avoid failure when `shadow-utils` is installed

KaneGreen, 2021-01-06:
I'm working on [let `passwd` using `busybox` to avoid failure when `shadow-utils` is installed](https://github.com/openwrt/luci/pull/4710) these days.

In this repo, if `autocore` is installed, the modification in that PR will not take effect. I hope the modification in that PR can be cherry-picked into this repo. (maybe when that PR is merged, need I PR here? or you will merge the "official" source?)

on the branch `master`:
There are two files need be modified.
```
package/lean/autocore/files/arm/rpcd_luci
package/lean/autocore/files/x86/rpcd_luci
```

1715173329, 2021-01-06:
We won't do anything until the PR you specified is merged.

1715173329, 2021-01-23:
Patch applied.

closed 2021-01-23 by 1715173329