immortalwrt/immortalwrt #253: 新版docker无法使用，强烈建议openwrt-18.06老系统换回旧版docker

xsmzhzy, 2021-01-10:
**昨天发现docker更新了，今天再次尝试了一下看能不能使用，我分别对openwrt-1806和openwrt-1907分支都进行了尝试，结果还是和上一版本一样，都不能正常使用。系统日志如下：**

Sun Jan 10 22:03:00 2021 daemon.info procd: - init complete -
Sun Jan 10 22:03:00 2021 user.info wsdd2[9148]: starting.
Sun Jan 10 22:03:01 2021 daemon.info dnsmasq[9339]: Connected to system UBus
Sun Jan 10 22:03:03 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:03.588453171Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Sun Jan 10 22:03:04 2021 daemon.err modprobe: failed to find a module named aufs
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.276917432Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.278053105Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.722094344Z" level=warning msg="Failed to find ip6tables: exec: \"ip6tables\": executable file not found in $PATH"
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.919809138Z" level=error msg="could not get initial namespace: no such file or directory"
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.932294702Z" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Sun Jan 10 22:03:04 2021 daemon.info modprobe: nf_conntrack is already loaded
Sun Jan 10 22:03:04 2021 daemon.info modprobe: nf_conntrack_netlink is already loaded
Sun Jan 10 22:03:04 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:04.943117582Z" level=error msg="failed to set to initial namespace, readlink /proc/8417/task/10265/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:03:05 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:05.220681863Z" level=error msg="failed to set to initial namespace, readlink /proc/8417/task/10265/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:03:05 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:05.397142591Z" level=error msg="failed to set to initial namespace, readlink /proc/8417/task/10265/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:03:05 2021 daemon.err dockerd[8417]: time="2021-01-10T14:03:05.397654140Z" level=error msg="failed to set to initial namespace, readlink /proc/8417/task/10265/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:03:42 2021 kern.notice kernel: [   63.148747] i915 0000:00:02.0: Failed to load DMC firmware i915/bxt_dmc_ver1_07.bin. Disabling runtime power management.
Sun Jan 10 22:03:42 2021 kern.notice kernel: [   63.159633] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
Sun Jan 10 22:03:54 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 10 22:03:54 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 10 22:03:58 2021 daemon.info dnsmasq[12368]: Connected to system UBus
Sun Jan 10 22:03:59 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 10 22:03:59 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 10 22:04:03 2021 daemon.info dnsmasq[12856]: Connected to system UBus
Sun Jan 10 22:04:24 2021 daemon.err modprobe: failed to find a module named aufs
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.213020742Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.213423394Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.268162109Z" level=warning msg="Failed to find ip6tables: exec: \"ip6tables\": executable file not found in $PATH"
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.574448139Z" level=error msg="could not get initial namespace: no such file or directory"
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.587633357Z" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Sun Jan 10 22:04:24 2021 daemon.info modprobe: nf_conntrack is already loaded
Sun Jan 10 22:04:24 2021 daemon.info modprobe: nf_conntrack_netlink is already loaded
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.598104022Z" level=error msg="failed to set to initial namespace, readlink /proc/13023/task/13029/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:04:24 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:24.820658407Z" level=error msg="failed to set to initial namespace, readlink /proc/13023/task/13029/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:04:25 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:25.035496239Z" level=error msg="failed to set to initial namespace, readlink /proc/13023/task/13029/ns/net: no such file or directory, initns fd -1: bad file descriptor"
Sun Jan 10 22:04:25 2021 daemon.err dockerd[13023]: time="2021-01-10T14:04:25.036436651Z" level=error msg="failed to set to initial namespace, readlink /proc/13023/task/13029/ns/net: no such file or directory, initns fd -1: bad file descriptor"

[image]

**最后，强烈建议大佬在新版docker没有稳定之前换回旧版docker。**

1715173329, 2021-01-10:
试过 master with dockerman 么？

xsmzhzy, 2021-01-10:
是的，dockerman一直在用

1715173329, 2021-01-10:
@xsmzhzy master + dockerman，不是单独一个 dockerman
master 下面能用么

xsmzhzy, 2021-01-10:
master貌似不稳定，所以没有尝试

1715173329, 2021-01-10:
@xsmzhzy 可以试试，毕竟新版本的 docker 是给 master 用的，官方可能没考虑过向后兼容（

xsmzhzy, 2021-01-10:
我也觉得是这样，不过最新的master+最新的docker=更多的不稳定，确实不想去尝试。

1715173329, 2021-01-10:
@xsmzhzy 那不一定

xsmzhzy, 2021-01-10:
谢谢天灵老大的热心指点，等哪天空下来我再试试。

1715173329, 2021-01-10:
就开着吧，最近 docker 问题多
看什么时候处理

hue715, 2021-01-11:
自从官方Master把docker-ce拆成docker+dockerd后，dockerman的依赖就缺dockerd。我自己补全依赖后，试了下从docker仓库安装adguardhome，可以正常加载。

xsmzhzy, 2021-01-11:
> 自从官方Master把docker-ce拆成docker+dockerd后，dockerman的依赖就缺dockerd。我自己补全依赖后，试了下从docker仓库安装adguardhome，可以正常加载。

我使用的是源码库自带的dockerman不存在依赖问题

1715173329, 2021-02-28:
试试选上 dockerd，官方设定的依赖有点奇怪。

xsmzhzy, 2021-03-05:
dockerd一直都是选上的

1715173329, 2021-03-06:
主要是 docker-ce 已经停止维护，不合适继续使用。
而且目前我们也无法得知具体是哪里出了问题。。。
这锅还得自己背，在官方的固件上面是正常运行的。。

xsmzhzy, 2021-03-06:
感谢天灵大佬的耐心解答！我现正在尝试使用openwrt-21.02分支，如果没有太大问题的话就在这个分支上跟下去了。

xsmzhzy, 2021-03-15:
是的，官方openwrt-21.02分支docker运行正常，immortalwrt/openwrt-21.02分支不能正常运行。

y9858, 2021-03-15:
> 是的，官方openwrt-21.02分支docker运行正常，immortalwrt/openwrt-21.02分支不能正常运行。

我这正常 难道是我姿势不对？

xsmzhzy, 2021-03-15:
真的是正常了吗？我是9天前尝试的，我自己查找下再试试

y9858, 2021-03-17:
> 真的是正常了吗？我是9天前尝试的，我自己查找下再试试

[image]
这样应该算正常吧？

xsmzhzy, 2021-03-17:
> > 是的，官方openwrt-21.02分支docker运行正常，immortalwrt / openwrt-21.02分支不能正常运行。
> 
> 我这正常难道是我姿势不对？
是的，确实是正常了

1715173329, 2021-03-17:
18.06 的问题还没解决呢，先别关，不然到时候就忘了

xsmzhzy, 2021-03-19:
> 18.06 的问题还没解决呢，先别关，不然到时候就忘了

好的，谢谢天灵大佬，你们辛苦了！

SuLingGG, 2021-03-19:
这里提供一个回退方式，经测试适用于 18.06-k5.4 分支，其他分支暂时未测试，也不做任何保证:

**但正如上文开发者所述，docker-ce 不再维护，使用者需自担风险。**

clone 源码，cd 至源码所在目录，先不要进行 update & install feeds 的操作，执行以下命令:

```
wget https://raw.githubusercontent.com/SuLingGG/OpenWrt-Mini/main/scripts/revert-docker.sh
chmod +x revert-docker.sh
./revert-docker.sh
```

然后进行 update & install feeds 的操作:

```
./scripts/feeds update -a && ./scripts/feeds install -a
```

接着 `make menuconfig`，**按顺序** 执行以下操作:

```
.
├── LuCI
│   └── Applications
│       └── luci-app-dockerman
├── Utilities
│   └── docker-ce
│       └── ...(全部选中 docker-ce 的所有特性)
└── Global build settings
    └── Kernel build options
        ├── -*- Enable kernel cgroups
        ├── ...
        ├── (选中 Enable kernel cgroups 到 Enable kernel namespaces 之间的所有项目)
        ├── ...
        └── -*- Enable kernel namespaces
```
然后正常编译，即可将 docker 回退至 docker-ce。

xsmzhzy, 2021-03-19:
这个脚本文件正是我渴望要的，感谢！

xsmzhzy, 2021-03-19:
暂时就用着docker-ce，等天灵大佬完美移植最新docker后再换，大佬们辛苦了！

yubanmeiqin9048, 2021-03-21:
先mark一下

M33FT, 2021-03-23:
Thanks to @xsmzhzy for opening this issue here
I have exactly the same problem
I tested it on different devices like RaspberryPi4B, RaspberryPi3B and Linksys WRT32X
and on all of them when I run a container it gives the following Error:

[image]

I do not have such a problem with the original version 19.0.7 Snapshot,
and when I add the dockerman to the feed and compile the ROM, everything works fine,
and docker and docker-compose all work. Only there I think the Docker version was 19.x.x but here it is 20.10.4.

[image]

[image]

I know you friends, especially @1715173329 are working hard on [immortalwrt](https://github.com/immortalwrt/immortalwrt) and this project is incredibly
complete and I have only been using this project since I got acquainted with it.

btw, if this problem is solved it will be great because I really need Docker.
Thanks guys for your strong support

279437541, 2021-04-09:
我的也是，能不能回到19.03.12版本的···

nameless3721, 2021-05-04:
内核缺少
CONFIG_KERNEL_NAMESPACES=y
CONFIG_KERNEL_UTS_NS=y
CONFIG_KERNEL_IPC_NS=y
CONFIG_KERNEL_USER_NS=y
CONFIG_KERNEL_PID_NS=y
CONFIG_KERNEL_NET_NS=y
添加后x64架构测试正常
[image]

1715173329, 2021-05-05:
> 内核缺少
> CONFIG_KERNEL_NAMESPACES=y
> CONFIG_KERNEL_UTS_NS=y
> CONFIG_KERNEL_IPC_NS=y
> CONFIG_KERNEL_USER_NS=y
> CONFIG_KERNEL_PID_NS=y
> CONFIG_KERNEL_NET_NS=y
> 添加后x64架构测试正常
> [image]

Thank you for test.

Ping @SuLingGG, could you please take a look on this?

SuLingGG, 2021-05-05:
> Thank you for test.
> 
> Ping @SuLingGG, could you please take a look on this?

Fantastic, I will try it later.

1715173329, 2021-05-05:
I finally understand why...
I have no idea, when `KERNEL_NAMESPACES` was set to be enabled by default.

On openwrt-19.07 or lower:
https://github.com/immortalwrt/immortalwrt/blob/fe3216610f050cf29a9b206c9c2a2a521c674b42/config/Config-kernel.in#L570
On openwrt-21.02 or higher:
https://github.com/immortalwrt/immortalwrt/blob/808e42c04b5c5d7955b1ce5d29f3e64069f14025/config/Config-kernel.in#L856

As this option was enabled by default, so even nobody recognized there is a missing dependency.
The reason why Lean's old package works, is in the package, `KERNEL_NAMESPACES` was added as a dependency.
https://github.com/coolsnowwolf/packages/blob/29e6c76ff35bb0573a5780b2fd51ab453b9f67fd/utils/docker-ce/Config.in#L8

1715173329, 2021-05-05:
Howerver, I would like to add it to docker's dependencies list, rather than enable it by default.

1715173329, 2021-05-05:
Hi guys, please pull to latest git commit and try again.

1715173329, 2021-05-06:
I'm going to close this as it was fixed in recent commit and tested by @SuLingGG.
Feel free to open new one if any issue remaining.

closed 2021-05-06 by 1715173329

linnvirgo, 2025-08-29:
> 试过 master with dockerman 么？

大佬能不能指点下详细点的方法啊，球球

linnvirgo, 2025-08-29:
> 自从官方Master把docker-ce拆成docker+dockerd后，dockerman的依赖就缺dockerd。我自己补全依赖后，试了下从docker仓库安装adguardhome，可以正常加载。

补哪些依赖啊，求大佬指点下