immortalwrt/immortalwrt #260: master 分支 dnsmasq-full/dnsmasq-2.83 编译失败

imoldchen, 2021-01-20:
openwrt/package/network/services/dnsmasq/patches/910-mini-ttl.patch
 openwrt/package/network/services/dnsmasq/patches/911-dnsmasq-filter-aaaa.patch
编译报错，然后根据提示修改了这两个补丁文件：
 #define LOPT_SCRIPT_TIME   360 下面源文件多了一行
 #define LOPT_PXE_VENDOR    361
修改后再编译，还是失败，报错：
```
make[4]: Entering directory '/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83'
make[5]: Entering directory '/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83/src'
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83=dnsmasq-2.83 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H    -DHAVE_DNSSEC   -DHAVE_CONNTRACK -DNO_ID     -DVERSION='"2.83"'      -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include    -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include   -c cache.c
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83=dnsmasq-2.83 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H    -DHAVE_DNSSEC   -DHAVE_CONNTRACK -DNO_ID     -DVERSION='"2.83"'      -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include    -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include   -c rfc1035.c
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83=dnsmasq-2.83 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H    -DHAVE_DNSSEC   -DHAVE_CONNTRACK -DNO_ID     -DVERSION='"2.83"'      -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include    -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include   -c util.c
ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83=dnsmasq-2.83 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/123456/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H    -DHAVE_DNSSEC   -DHAVE_CONNTRACK -DNO_ID     -DVERSION='"2.83"'      -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include    -I/home/123456/openwrt/staging_dir/target-x86_64_musl/usr/include   -c option.c
option.c: In function 'one_opt':
option.c:3953:5: error: duplicate case value
     case LOPT_PXE_VENDOR: /* --dhcp-pxe-vendor */
     ^~~~
option.c:2832:5: note: previously used here
     case LOPT_MINTTL: /* --min-ttl */
     ^~~~
make[5]: *** [/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83/Makefile:164: option.o] Error 1
make[5]: Leaving directory '/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83/src'
make[4]: *** [Makefile:89: all] Error 2
make[4]: Leaving directory '/home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83'
make[3]: *** [Makefile:201: /home/123456/openwrt/build_dir/target-x86_64_musl/dnsmasq-full/dnsmasq-2.83/.built] Error 2
make[3]: Leaving directory '/home/123456/openwrt/package/network/services/dnsmasq'
time: package/network/services/dnsmasq/full/compile#0.49#0.17#2.39
    ERROR: package/network/services/dnsmasq failed to build (build variant: full).
make[2]: *** [package/Makefile:114: package/network/services/dnsmasq/compile] Error 1
make[2]: Leaving directory '/home/123456/openwrt'
make[1]: *** [package/Makefile:108: /home/123456/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/123456/openwrt'
make: *** [/home/123456/openwrt/include/toplevel.mk:234: world] Error 2
```

1715173329, 2021-01-20:
Cc: @AmadeusGhost @ElonH

AmadeusGhost, 2021-01-20:
`LOPT_MINTTL` 应改为 362，稍等我跑下编译看看

imoldchen, 2021-01-20:
> 
> 
> `LOPT_MINTTL` 应改为 362，稍等我跑下编译看看

911-dnsmasq-filter-aaaa.patch 添加的是 #define LOPT_FILTER_AAAA   362

AmadeusGhost, 2021-01-20:
> 911-dnsmasq-filter-aaaa.patch 添加的是 #define LOPT_FILTER_AAAA 362

那就改为 363 呗，这个只要不重复就行了

ElonH, 2021-01-20:
`LOPT_MINTTL` 和 `LOPT_PXE_VENDOR` 定义的值一样了，试着改成不同值

imoldchen, 2021-01-20:
> > 911-dnsmasq-filter-aaaa.patch 添加的是 #define LOPT_FILTER_AAAA 362
> 
> 那就改为 363 呗，这个只要不重复就行了

我的意思是数字没有重复，patch文件里没有361，而源文件里有。

ElonH, 2021-01-20:
> > > 911-dnsmasq-filter-aaaa.patch 添加的是 #define LOPT_FILTER_AAAA 362
> > 
> > 那就改为 363 呗，这个只要不重复就行了
> 
> 我的意思是数字没有重复，patch文件里没有361，而源文件里有。

https://github.com/project-openwrt/openwrt/blob/3a7d5224cae06eef0a5a36a1b2aaa1815db0487a/package/network/services/dnsmasq/patches/910-mini-ttl.patch#L20

AmadeusGhost, 2021-01-20:
Just run `git pull` and it will works fine.

commit 626b9f23 (2021-01-20): dnsmasq: refresh patches

Fix: #260
commit 8bd39732 (2021-01-21): dnsmasq: refresh patches

Fix: #260
closed 2021-01-20 by 1715173329