immortalwrt/immortalwrt #274: openwrt-18.06-k5.4 这个版本的源码编译失败

wyap-echo, 2021-02-02:
机型是R2S 使用的是 openwrt-18.06-k5.4 这个版本的源码，是因为这两个选项冲突了吗

< > kmod-fast-classifier................... Kernel driver for FAST Classifier    
 -*- kmod-fast-classifier-noload     

* check_data_file_clashes: Package kmod-fast-classifier-noload wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/lib/modules/5.4.93/fast-classifier.ko
	But that file is already provided by package  * kmod-fast-classifier
 * opkg_install_cmd: Cannot install package kmod-fast-classifier-noload.
 * check_data_file_clashes: Package kmod-fast-classifier-noload wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/lib/modules/5.4.93/fast-classifier.ko
	But that file is already provided by package  * kmod-fast-classifier
 * opkg_install_cmd: Cannot install package luci-app-turboacc.

AmadeusGhost, 2021-02-02:
选 kmod-fast-classifier 就行

closed 2021-02-02 by AmadeusGhost

wyap-echo, 2021-02-02:
> 选 kmod-fast-classifier 就行

奇怪这个 kmod-fast-classifier-noload 是默认选择的 我没办法手动去掉 有什么办法去掉吗

AmadeusGhost, 2021-02-02:
> 奇怪这个 kmod-fast-classifier-noload 是默认选择的 我没办法手动去掉 有什么办法去掉吗

找到了，是 https://github.com/project-openwrt/openwrt/commit/4201dbd9bb43177511163c0c19aba50e308feb5a 的错误，稍等我修正一下

AmadeusGhost, 2021-02-02:
> 奇怪这个 kmod-fast-classifier-noload 是默认选择的 我没办法手动去掉 有什么办法去掉吗

Fixed in https://github.com/project-openwrt/openwrt/commit/381764db1de645ab089734fb347550de474e5615

wyap-echo, 2021-02-02:
谢谢 不过这个openwrt-18.06-k5.4分支的的源码好像还没更新

1715173329, 2021-02-02:
这是你自己没有更新配置文件导致的
最快的解决方案是删除配置文件重新配置