immortalwrt/immortalwrt #279: ipv6没有显示

osxdiy, 2021-02-08:
IPV6正常但是概况是无显示
[image]

1715173329, 2021-02-08:
什么叫做正常？
请出示你的接口页面截图

AmadeusGhost, 2021-02-08:
代码逻辑是这样的：
```
<% if has_ipv6 then %>
var si6 = document.getElementById('wan6_i');
var ss6 = document.getElementById('wan6_s');
var ifc6 = info.wan6;

if (ifc6 && ifc6.ifname && ifc6.proto != 'none')
{
	xxx
else
{
	si6.innerHTML = '<img src="<%=resource%>/icons/ethernet_disabled.png" /><br /><small>?</small>';
	ss6.innerHTML = '<em><%:Not connected%></em>';
}
<% end %>
```
个人不觉得你的 ipv6 正常

osxdiy, 2021-02-08:
我的IPV6是正常的，就是概况中IPV6 WAN无显示，是不是多拨造成的

osxdiy, 2021-02-08:
[image]
[image]
接口如上

AmadeusGhost, 2021-02-08:
wan6 没有拿到 ipv6 地址，自然不会显示。

osxdiy, 2021-02-08:
用LEAN的原码编译的WAN6没有IPV6地址，但是IPV6 WAN 是显示的

AmadeusGhost, 2021-02-08:
wan 或者 wan6 拿到都可以，应该是多拨搞乱的。

osxdiy, 2021-02-08:
谢谢了

closed 2021-02-08 by osxdiy

1715173329, 2021-02-08:
LuCI 默认从第一个防火墙为 `WAN` 的接口获取 IP 地址信息
按照图中所示，你的 `VWAN1` 没有有效 IPv6 信息，故不会显示

osxdiy, 2021-02-08:
谢谢你的解答