immortalwrt/immortalwrt #280: ssr+订阅报错

zheshifandian, 2021-02-08:
2021-02-09 07:44:14 重启服务成功
2021-02-09 07:44:14 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:512: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:413: in function </usr/share/shadowsocksr/subscribe.lua:365>
	/usr/share/shadowsocksr/subscribe.lua:126: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:514: in function </usr/share/shadowsocksr/subscribe.lua:512>
2021-02-09 07:44:14 stack traceback:
2021-02-09 07:44:14 /usr/share/shadowsocksr/subscribe.lua:126: attempt to index local 'info' (a nil value)
2021-02-09 07:44:12 成功解析节点数量: 43

1715173329, 2021-02-09:
提供你的订阅链接，发送到 cnsztl@project-openwrt.eu.org

zheshifandian, 2021-02-11:
> 提供你的订阅链接，发送到 [cnsztl@project-openwrt.eu.org](mailto:cnsztl@project-openwrt.eu.org)

发了

1715173329, 2021-02-11:
> 发了

>11.02.2021, 21:34, "Jack Sun" <su\*21@gmail.com>:
> https://\*.xyz/link/\*bW\*?sub=3

该订阅为 V2RayN 标准，经测试正常获取节点内容，
如有异常，请更新你的 SSRPlus 版本

> https://\*.xyz/link/\*bW\*?list=shadowrocket

该订阅为 Shadowrocket 标准，无法通过 SSRPlus 进行订阅

closed 2021-02-11 by 1715173329