immortalwrt/immortalwrt #284: 18.06分支报错

281677160, 2021-02-16:
* pkg_hash_fetch_best_installation_candidate: Packages for libreadline8 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-autoupdate:
 * 	libncursesw6
 * opkg_install_cmd: Cannot install package luci-app-autoupdate.
 * pkg_hash_fetch_best_installation_candidate: Packages for fdisk found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package fdisk.
make[2]: *** [package/Makefile:68: package/install] Error 255

一直编译都没问题的，早上还编译也没问题的，晚上编译就报错了，请问一下 这个怎么处理，感谢！

1715173329, 2021-02-16:
This is a duplicate of #283.

281677160, 2021-02-16:
感谢回复，那就只能等了，定时更新这个插件不编译看看能成功不

xiaobaishu1, 2021-02-17:
> This is a duplicate of #283.

[image]

closed 2021-02-17 by 1715173329