immortalwrt/immortalwrt #286: rtl8821cu 编译失败

garypang13, 2021-02-19:
```
make[2]: Leaving directory '/home/garypang/sdk/package/kernel/linux'
time: package/kernel/linux/compile#1.77#0.26#1.99
make[2]: Entering directory '/home/garypang/sdk/feeds/custom/rtl8821cu'
mkdir -p /home/garypang/sdk/dl
SHELL= flock /home/garypang/sdk/tmp/.rtl8821cu-2020-12-19-428a0820.tar.xz.flock -c '    /home/garypang/sdk/scripts/download.pl "/home/garypang/sdk/dl" "rtl8821cu-2020-12-19-428a0820.tar.xz" "77958d3bff8b0145504a10959765be0e3743b9c4880a5173d156238c2c569a56" "" || (  /home/garypang/sdk/scripts/dl_github_archive.py --dl-dir="/home/garypang/sdk/dl" --url="https://github.com/brektrou/rtl8821CU.git" --version="428a0820487418ec69c0edb91726d1cf19763b1e" --subdir="rtl8821cu-2020-12-19-428a0820" --source="rtl8821cu-2020-12-19-428a0820.tar.xz" --hash="77958d3bff8b0145504a10959765be0e3743b9c4880a5173d156238c2c569a56" || (       echo "Checking out files from the git repository..."; mkdir -p /home/garypang/sdk/tmp/dl && cd /home/garypang/sdk/tmp/dl && rm -rf rtl8821cu-2020-12-19-428a0820 && [ \! -d rtl8821cu-2020-12-19-428a0820 ] && git clone  https://github.com/brektrou/rtl8821CU.git rtl8821cu-2020-12-19-428a0820 && (cd rtl8821cu-2020-12-19-428a0820 && git checkout 428a0820487418ec69c0edb91726d1cf19763b1e ; git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd rtl8821cu-2020-12-19-428a0820 && git log -1 --format='\''@%ct'\''` && rm -rf rtl8821cu-2020-12-19-428a0820/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c rtl8821cu-2020-12-19-428a0820 |    xz -zc -7e > /home/garypang/sdk/tmp/dl/rtl8821cu-2020-12-19-428a0820.tar.xz && mv /home/garypang/sdk/tmp/dl/rtl8821cu-2020-12-19-428a0820.tar.xz /home/garypang/sdk/dl/ && rm -rf rtl8821cu-2020-12-19-428a0820; );  )    '
make[2]: *** No rule to make target '/home/garypang/sdk/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/backport/autoconf.h', needed by '/home/garypang/sdk/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[2]: Leaving directory '/home/garypang/sdk/feeds/custom/rtl8821cu'
```

AmadeusGhost, 2021-02-19:
kernel package 应该不能在 sdk 下编译吧

garypang13, 2021-02-19:
不好意思,上面的错误是环境的问题, 我固件编译错了,然后在sdk中测试了下,

实际的错误如下:
```
CC [M]  /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/hal/btc/halbtc8821c1ant.o
  CC [M]  /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/hal/btc/halbtc8821c2ant.o
  CC [M]  /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/platform/platform_ops.o
  CC [M]  /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mp.o
/home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mp.c: In function 'rtfloor':
/home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mp.c:33:6: error: SSE register return with SSE disabled
  int i = x - 2;
      ^
make[4]: *** [scripts/Makefile.build:262: /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mp.o] Error 1
make[3]: *** [Makefile:1726: /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820] Error 2
make[3]: Leaving directory '/home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.99'
make[2]: *** [Makefile:85: /home/garypang/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/.built] Error 2
make[2]: Leaving directory '/home/garypang/openwrt/feeds/custom/rtl8821cu'
time: package/feeds/custom/rtl8821cu/compile#147.29#19.23#170.17
    ERROR: package/feeds/custom/rtl8821cu failed to build.
make[1]: *** [package/Makefile:114: package/feeds/custom/rtl8821cu/compile] Error 1
```

1715173329, 2021-02-19:
Try to apply this patch:
```patch
diff --git a/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch b/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
index 8d33720aa1..bc753af413 100644
--- a/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
+++ b/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
@@ -1,18 +1,11 @@
CN_SZTL@CN-SZTL-DEV:~/project/lede$ nano package/ctcgfw/rtl8821cu/Makefile
CN_SZTL@CN-SZTL-DEV:~/project/lede$ git diff
diff --git a/package/ctcgfw/rtl8821cu/Makefile b/package/ctcgfw/rtl8821cu/Makefile
index fed7bf4d41..fa22f3f2d3 100644
--- a/package/ctcgfw/rtl8821cu/Makefile
+++ b/package/ctcgfw/rtl8821cu/Makefile
@@ -55,6 +55,10 @@ ifeq ($(ARCH),aarch64)
        EXTRA_KCONFIG += CONFIG_MP_VHT_HW_TX_MODE=n
 endif
 
+ifeq ($(ARCH),x86_64)
+       EXTRA_CFLAGS += -mhard-float
+endif
+
 EXTRA_CFLAGS:= \
        -DRTW_SINGLE_WIPHY \
        -DRTW_USE_CFG80211_STA_EVENT \
```

garypang13, 2021-02-19:
> Try to apply this patch:
> 
> ```diff
> diff --git a/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch b/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
> index 8d33720aa1..bc753af413 100644
> --- a/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
> +++ b/package/ctcgfw/rtl8821cu/patches/010-remove-extra-cflags.patch
> @@ -1,18 +1,11 @@
> CN_SZTL@CN-SZTL-DEV:~/project/lede$ nano package/ctcgfw/rtl8821cu/Makefile
> CN_SZTL@CN-SZTL-DEV:~/project/lede$ git diff
> diff --git a/package/ctcgfw/rtl8821cu/Makefile b/package/ctcgfw/rtl8821cu/Makefile
> index fed7bf4d41..fa22f3f2d3 100644
> --- a/package/ctcgfw/rtl8821cu/Makefile
> +++ b/package/ctcgfw/rtl8821cu/Makefile
> @@ -55,6 +55,10 @@ ifeq ($(ARCH),aarch64)
>         EXTRA_KCONFIG += CONFIG_MP_VHT_HW_TX_MODE=n
>  endif
>  
> +ifeq ($(ARCH),x86_64)
> +       EXTRA_CFLAGS += -mhard-float
> +endif
> +
>  EXTRA_CFLAGS:= \
>         -DRTW_SINGLE_WIPHY \
>         -DRTW_USE_CFG80211_STA_EVENT \
> ```

不行,还是报一样的错误.

closed 2021-02-19 by 1715173329

AmadeusGhost, 2021-02-19:
Just run `git pull` and `make package/rtl8821cu/{clean,compile} V=s`.

commit b6d96714 (2021-02-19): rtl8821cu: fix build on x86 platform

Fix: #286
commit 75d4e97c (2021-02-19): rtl8821cu: fix build on x86 platform

Fix: #286
commit 83755a65 (2021-02-19): rtl8821cu: fix build on x86 platform

Fix: #286