immortalwrt/immortalwrt #288: qBittorrent-Enhanced-Edition编译错误

eggydutch, 2021-02-21:
openwrt-21.02分支下，qBittorrent-Enhanced-Edition编译错误（编译固件为R2S）。
[image]

garypang13, 2021-02-23:
编译qBittorrent花费的时间占用的空间失败的概率都太大了,不如直接下载作者已编译好的静态二进制.
https://github.com/garypang13/openwrt-static-qb/

1715173329, 2021-02-28:
检查下 boost-system 这个依赖项有没有被选中

hue715, 2021-02-28:
> 检查下 boost-system 这个依赖项有没有被选中

Master分支下，确认boost-system依赖项已选中，但编译qBittorrent仍会报相同错误。

1715173329, 2021-03-11:
更新了一下，再试试，不要用云编译

xcyll, 2021-03-24:
> 更新了一下，再试试，不要用云编译

21.02分支 transmission编译成功 但不能启用 方框下拉无选项
[image]
[image]

ricksuzade-maker, 2021-04-15:
可否更换成官方版本，这个所谓的增强版会被很多pt网站直接block掉，没法用

zxlhhyccc, 2021-04-15:
> openwrt-21.02分支下，qBittorrent-Enhanced-Edition编译错误（编译固件为R2S）。
> [image]

这个情况是libtorrent-rasterbar的版本问题，你使用的肯定是默认官方的2.x版本，请改为1.12.x版本编译。。。。

1715173329, 2021-04-18:
> 可否更换成官方版本，这个所谓的增强版会被很多pt网站直接block掉，没法用

我觉得这个有理
实际上我们可以做成两个版本，可以相互切换。

eggydutch, 2021-05-08:
> > openwrt-21.02分支下，qBittorrent-Enhanced-Edition编译错误（编译固件为R2S）。
> > [image]
> 
> 这个情况是libtorrent-rasterbar的版本问题，你使用的肯定是默认官方的2.x版本，请改为1.12.x版本编译。。。。

谢谢你的回复及建议，因为我不懂具体如何操作，已经换回OP 18.06的版本了。最近两天又尝试了编译一下21.02，qB仍然是出错

1715173329, 2021-05-08:
该问题已确定并复现。
~~暂时不知道如何修复~~

eggydutch, 2021-05-08:
> 更新了一下，再试试，不要用云编译

对不起大佬我只有云编译的环境

xcyll, 2021-05-11:
可以用transmission 保种好 不过21.02分支下也有些问题 openwrt issue区提到的解决方案
[image]

xcyll, 2021-05-20:
> 可以用transmission 保种好 不过21.02分支下也有些问题
> [image]
> openwrt issue区提到的解决方案[openwrt/luci#4697](https://github.com/openwrt/luci/issues/4697)
> [image]
✨✨关于transmission3.0✨✨
最新源码解决了 可以了 目前还遗留一个问题：首次启动TR3.0 不能使用【不出现 打开web管理界面按钮】刷新后也不出现  登录op后台 系统-启动项-重启transmission才可用

xcyll, 2021-05-20:
可以用transmission 保种好 不过21.02分支下也有些问题
✨✨关于transmission3.0✨✨
最新源码【5/19】解决了运行权限不能勾选的问题
目前还遗留一个问题：首次启动TR3.0 不能使用【不出现 打开web管理界面按钮】刷新后也不出现 登录op后台 系统-启动项-重启transmission才可用