immortalwrt/immortalwrt #291: baiduPCS-web rice

juebanlin, 2021-02-26:


AmadeusGhost, 2021-02-26:
试试这样改下可不可以用：
```
--- a/package/lean/baidupcs-web/Makefile
+++ b/package/lean/baidupcs-web/Makefile
@@ -43,6 +43,15 @@ define Package/$(PKG_NAME)/description
 BaiduPCS-Web is a web controller for BaiduPCS-Go
 endef
 
+define Build/Compile
+( \
+  GOOS=$$$$(go env GOOS) GOARCH=$$$$(go env GOARCH) go get -v github.com/GeertJohan/go.rice/rice/... ; \
+  cd $(PKG_BUILD_DIR)/internal/pcsweb ; \
+  "$$$$(go env GOPATH)/bin/rice" embed-go ; \
+  $(call GoPackage/Build/Compile) ; \
+)
+endef
+
 define Package/$(PKG_NAME)/install
 	$(call GoPackage/Package/Install/Bin,$(PKG_INSTALL_DIR))
 	$(INSTALL_DIR) $(1)/usr/bin
```

fuqiang03, 2021-02-26:
[image]
我也贴个图 自从升级了go1.16 貌似出现了问题

juebanlin, 2021-02-26:
有2种方法，编译package时安装rice，
go get github.com/GeertJohan/go.rice
go get github.com/GeertJohan/go.rice/rice
export PATH=$GOPATH/bin:$PATH
或者在操作系统安装rice支持
sudo -E apt-get -qq install -y golang-rice

zxlhhyccc, 2021-02-27:
> ide编译这个插件项目时,在build.sh加入一下代码可以编译成功，在makefile里面就不知道怎么搞了,毕竟引用的golang不是全局的
> go get github.com/GeertJohan/go.rice
> go get github.com/GeertJohan/go.rice/rice
> export PATH=$GOPATH/bin:$PATH
通过Makefile编译的二进制文件大小是12+m，而直接二进制文件大小是20+m，使用Makefile编译的无法启动运行。。。

commit 03bc1d2c (2021-03-01): baidupcs-web: add missing rice dependency

Fix: #291
commit 2a15075c (2021-03-01): baidupcs-web: add missing rice dependency

Fix: #291
closed 2021-02-28 by 1715173329