immortalwrt/immortalwrt #296: 编译时空间占用太大，Gzip压缩镜像体积不缩小

LoogCP, 2021-03-04:
编译等21.02分支，因为之前编译lean固件时空间设置很大（8G左右），所以这边也这样尝试，但是占用空间太大了，只选择了输出immortalwrt-x86-64-generic-squashfs-combined.img.gz、immortalwrt-x86-64-generic-squashfs-rootfs.img.gz和一个vdi，但是每个都是9G左右，导致空间占用太大，编译失败
而且镜像输出目录是build_dir/target-x86_64_musl/linux-x86_64/tmp，而不是在bin里面，不知道是不是因为空间不够没有复制过去
```
$du immortalwrt/ -h -d 1
50M	immortalwrt/target
3.0M	immortalwrt/tools
3.0G	immortalwrt/dl
2.2M	immortalwrt/scripts
380K	immortalwrt/include
284M	immortalwrt/bin
35G	immortalwrt/build_dir
194M	immortalwrt/feeds
345M	immortalwrt/.git
1.1G	immortalwrt/.ccache
726M	immortalwrt/tmp
148M	immortalwrt/package
776K	immortalwrt/toolchain
68K	immortalwrt/config
100K	immortalwrt/LICENSES
3.6G	immortalwrt/staging_dir
45G	immortalwrt/
```
这个还是删除了9G等vdi镜像的空间占用（删除vdi是因为这个目录里面生成等vdi文件在virtualbox里面打开失败）

另外，luci-app-gowebdav和luci-app-qbittorrent这两个软件选上会编译失败

klever1988, 2021-03-05:
把编译目录放在btrfs格式的分区 启用透明压缩 可以控制到10g之内

LoogCP, 2021-03-05:
我没有btrfs格式的分区，我/是jfs，/home是ext4格式，而且我也不知道透明压缩怎么开启
我现在准备把lean的lede删掉释放空间再来试试

LoogCP, 2021-03-05:
把lean大的lede删了之后编译，倒是编译成功了，不过lean大佬和这个等配置不兼容，删除的时候忘记备份固件了，所以还没直接刷上机器，配置迁移需要太多精力
另外就是lean大佬的nginx可以选择不含ssl的nginx，但是这个不行，然而用含ssl的nginx的话，无法通过frp远程访问路由器管理界面，所以是真的纠结导致要不要换成这个

closed 2021-03-05 by LoogCP