immortalwrt/immortalwrt #299: How do I enable wifi?

dausruddin, 2021-03-04:
Hi. I am trying to enable wifi with my RTL8811CU USB adapter with no luck. I have selected rtl8821cu in `make menuconfig`, but Wireless menu does not show up.
[image]

 Is there any additional packages I need to enable?

I am trying with branch 21.02.

Thanks

AmadeusGhost, 2021-03-05:
> Is there any additional packages I need to enable?

Wpad and iwinfo are also necessary packages for wireless. Restart after plugging in the wireless network card. If it still does not appear, please send the kernel log and lsusb results.

dausruddin, 2021-03-05:
I noticed something wrong. The adapter is detected as `Realtek DISK`
```root@ImmortalWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux 5.4.99 ohci_hcd Generic Platform OHCI controller
Bus 004 Device 001: ID 1d6b:0003 Linux 5.4.99 xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 0bda:1a2b Realtek DISK
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.99 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 5.4.99 xhci-hcd xHCI Host Controller
```

This is the image of the device. 
[image]

- Kernel log: [Log.txt](https://github.com/immortalwrt/immortalwrt/files/6088020/Log.txt)
- .config: [config.txt](https://github.com/immortalwrt/immortalwrt/files/6088024/config.txt)

AmadeusGhost, 2021-03-05:
> Bus 001 Device 003: ID 0bda:1a2b Realtek DISK

It seems that usb-modeswitch is also needed to switch the usb mode, maybe you can find it on the search engine.

dausruddin, 2021-03-09:
Sorry for the delay. I installed [usb-modeswitch](https://openwrt.org/packages/pkgdata/usb-modeswitch) and it is working great now.

Thanks a lot for your help :)

closed 2021-03-09 by dausruddin

kzh1980, 2022-02-03:
> Sorry for the delay. I installed [usb-modeswitch](https://openwrt.org/packages/pkgdata/usb-modeswitch) and it is working great now.
> 
> Thanks a lot for your help :)

it seems that i am having the same one. how do you set the WIFI setting of this device? could you please share yours?

this is my seetings, it didn't works 
Client mode,
AC, 5G,  149,  RU
CCMP force

if the seeting is corrent, the WIFI should be working under Master mode.

dausruddin, 2022-02-04:
Hi @kzh1980. It has been a while and I no longer using usb wifi adapter. Now I am using a router. I can't remember sadly, but I'm sure that it didn't work when I turned on encryption (enabling password). It only work when I set to no encryption (no password).

I have this image that I captured back then.

<kbd>[image]</kbd>

And this is what my friend sent me, also long time ago.

<kbd>[image]</kbd>

Sorry I couldn't help much.