immortalwrt/immortalwrt #309: R21.02 分支 R4S 启动后LAN口不工作

wyap-echo, 2021-03-10:
通电启动后LAN灯不亮 wan口接上级路由可以Ping通 但无法进入后台 ssh也无法连接（这个只有把防火墙里面 wan 口的入站数据 改为接受才可以访问），重新编译并替换防火墙文件后可以进入后台，但是lan 口依旧无法访问显示设备不存在
[image]
下面是系统日志
`Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Fri Jan 18 16:50:20 2013 kern.notice kernel: [    0.000000] Linux version 5.4.102 (action@debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r18194-4deeb05e91)) #0 SMP PREEMPT Wed Mar 10 09:16:27 2021
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Machine model: FriendlyElec NanoPi R4S
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000] On node 0 totalpages: 261632
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000]   DMA32 zone: 4088 pages used for memmap
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s45976 r8192 d31848 u86016
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000] pcpu-alloc: s45976 r8192 d31848 u86016 alloc=21*4096
Fri Jan 18 16:50:20 2013 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] CPU features: detected: GIC system register CPU interface
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Fri Jan 18 16:50:20 2013 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff1a0000 root=PARTUUID=5452574f-02 rw rootwait
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] Memory: 1004052K/1046528K available (7870K kernel code, 576K rwdata, 2440K rodata, 1728K init, 729K bss, 34284K reserved, 8192K cma-reserved)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] 	Tasks RCU enabled.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: 256 SPIs implemented
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: 0 Extended SPIs implemented
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: Distributor has no Range Selector support
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: 16 PPIs implemented
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: no VLPI support, no direct LPI support
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @3dc80000 (flat, esz 8, psz 64K, shr 0)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] ITS: using cache flushing for cmd queue
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: using LPI property table @0x000000003dc40000
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GIC: using cache flushing for LPI property table
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: CPU0: using allocated LPI pending table @0x000000003dc50000
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
Fri Jan 18 16:50:20 2013 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x39c with crng_init=0
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.001879] Console: colour dummy device 80x25
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.002352] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.003331] pid_max: default: 32768 minimum: 301
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.003971] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.004680] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.007372] ASID allocator initialised with 32768 entries
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.007999] rcu: Hierarchical SRCU implementation.
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.008733] Platform MSI: interrupt-controller@fee20000 domain created
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.009723] PCI/MSI: /interrupt-controller@fee00000/interrupt-controller@fee20000 domain created
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.011180] smp: Bringing up secondary CPUs ...
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.012181] Detected VIPT I-cache on CPU1
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.012217] GICv3: CPU1: found redistributor 1 region 0:0x00000000fef20000
Fri Jan 18 16:50:20 2013 kern.info kernel: [    0.012233] GICv3: CPU1: using allocated LPI pending table @0x000000003dc60000
[... 470 lines cut ...]
Wed Mar 10 23:41:57 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar 10 23:41:57 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar 10 23:41:57 2021 daemon.info dnsmasq[5844]: Connected to system UBus
Wed Mar 10 23:42:03 2021 kern.info kernel: [   31.715718] vcc1v8_s3: supplied by vcc_1v8
Wed Mar 10 23:42:03 2021 kern.info kernel: [   31.716219] vbus_typec: disabling
Wed Mar 10 23:42:06 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0)
[... cut ...]
Wed Mar 10 23:42:20 2021 daemon.err nmbd[3578]:
Wed Mar 10 23:42:20 2021 daemon.err nmbd[3578]:   Samba name server IMMORTALWRT is now a local master browser for workgroup WORKGROUP on subnet 192.168.31.127
Wed Mar 10 23:42:20 2021 daemon.err nmbd[3578]:
Wed Mar 10 23:42:20 2021 daemon.err nmbd[3578]:   *****
Wed Mar 10 23:42:23 2021 cron.err crond[3478]: time disparity of 4281531 minutes detected
Wed Mar 10 23:45:02 2021 daemon.err uhttpd[1963]: luci: accepted login on / for root from 192.168.31.57`

1715173329, 2021-03-11:
运行 `make defconfig`，贴出你的配置

wyap-echo, 2021-03-11:
谢谢大佬 是配置问题 我重新配置了一遍 编译后就正常了

closed 2021-03-11 by wyap-echo

Xyrelln, 2022-07-04:
> 谢谢大佬 是配置问题 我重新配置了一遍 编译后就正常了

Could you share how you fixed this? Same issue here. Thank you.