immortalwrt/immortalwrt #313: amadeusghost

i3roly, 2021-03-14:
@amadeusghost 我了解遵守保密协议的含义。 我的问题是我可以在哪里获得修订版6.0.2.0。 显然，如果我公开发布修订版本，我希望能被起诉。 我只想要我自己的固件。 我没有发布固件的源代码

1715173329, 2021-03-14:
Talk it in English.

i3roly, 2021-03-14:
@1715173329 @AmadeusGhost i know what you're saying that 6.0.2.0 is closed-source. i want to join the developer program where i can access this source. i do not have any plans to release the source publicly. right now i have a custom build DD-WRT for DIR-882. i only release the binary. i do not provide the source. i have good results with the 5.0.4.0 driver but i want the newer version. PMF support is not complete in 5.x and there are many RX errors because of that. also, i have noticed poor 2.4GHZ performance.

i assume there would be access to updated hw_nat source as well? i would like this too.

AmadeusGhost, 2021-03-14:
我前面说过了，我们手里都没有您所说的 mtwifi 6.0.2.0 代码。
As I said before, none of us have the mtwifi 6.0.2.0 source you mentioned.

i3roly, 2021-03-14:
oh, i thought you meant that if you did have it, you couldn't share it. how does anyone get access? the russians have access. they seem to have the the best access. i want access too! don't the guys at right.com.cn have newer stuff? i can't read right.com.cn like i was able to a few years ago, too many loginchecks and i am not a native chinese citizen so i can't make an account without paying through baidu

AmadeusGhost, 2021-03-14:
Nope, we don't have it...

closed 2021-03-14 by AmadeusGhost

i3roly, 2021-03-14:
ah shit, thanks buddies. 

well, at least i made some new friends (or so i hope ^_^).

have any of you tried out @ptpt52's netflow with the MTK ethernet driver? have you had good results? i couldn't get it to work. i am thinking about giving it another try soon.

AmadeusGhost, 2021-03-14:
Natflow can only be used with the open source mt76 driver. At least for now, I'm busy with my studies and no time to try it.

i3roly, 2021-03-14:
i had no idea it couldn't be used with the mt_wifi driver.

i should ask @Ptpt52 about this. i was under the impression that either the mt76 or the proprietary mediatek sources would identify the wireless interface as an 'extif' (external interface) and thus route. 

@ptpt52 PING (i am sure you're a very busy man, but your input would be welcome here. i will also ask on your github).

i feel like i'm missing out on all the good things in the mediatek enthusiast community. the mt76 driver sucks, have you had as poor of an experience as i have? even with the warts of 5.0.4.0, it is much, much better than the mt76 version i've tried (pretty recent version, as late as mid january).