immortalwrt/immortalwrt #319: 您好，编译R2s报错

blacknesswing, 2021-03-15:
您好 ：用P3TERX Actions-OpenWrt编译18.06-k54报错如下，请问这个咋办？谢谢
make[5]: *** [/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj.host/icutools/deps/icu-small/source/i18n/number_longnames.o] Error 1
make[4]: *** [node] Error 2
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/.built] Error 2
make[2]: *** [package/feeds/packages/node/compile] Error 1
tools/icu/icutools.host.mk:563: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj.host/icutools/deps/icu-small/source/i18n/number_longnames.o' failed
rm 96141353e73028c0b03b1675eb32725ef90e6ccd.intermediate cf2418fc26caf22d819d84e91b6166a8c07940c2.intermediate
Makefile:104: recipe for target 'node' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0'
Makefile:195: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#9.97#0.48#10.38
    ERROR: package/feeds/packages/node failed to build.
package/Makefile:112: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

AmadeusGhost, 2021-03-15:
日志不够详细，要 Error 1 上面那段的具体报错

blacknesswing, 2021-03-15:
> 日志不够详细，要 Error 1 上面那段的具体报错

谢谢大佬，我又复制了，您看看
LD_LIBRARY_PATH=/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/lib.host:/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated/../../deps/v8/test/torque /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated/../../deps/v8/src/builtins /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated/../../deps/v8/src/objects /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated/../../deps/v8/third_party/v8/builtins; "/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/torque" ../../deps/v8/src/builtins/arguments.tq ../../deps/v8/src/builtins/array-copywithin.tq ../../deps/v8/src/builtins/array-every.tq ../../deps/v8/src/builtins/array-filter.tq ../../deps/v8/src/builtins/array-find.tq ../../deps/v8/src/builtins/array-findindex.tq ../../deps/v8/src/builtins/array-foreach.tq ../../deps/v8/src/builtins/array-join.tq ../../deps/v8/src/builtins/array-lastindexof.tq ../../deps/v8/src/builtins/array-map.tq ../../deps/v8/src/builtins/array-of.tq ../../deps/v8/src/builtins/array-reduce-right.tq ../../deps/v8/src/builtins/array-reduce.tq ../../deps/v8/src/builtins/array-reverse.tq ../../deps/v8/src/builtins/array-shift.tq ../../deps/v8/src/builtins/array-slice.tq ../../deps/v8/src/builtins/array-some.tq ../../deps/v8/src/builtins/array-splice.tq ../../deps/v8/src/builtins/array-unshift.tq ../../deps/v8/src/builtins/array.tq ../../deps/v8/src/builtins/base.tq ../../deps/v8/src/builtins/bigint.tq ../../deps/v8/src/builtins/boolean.tq ../../deps/v8/src/builtins/collections.tq ../../deps/v8/src/builtins/data-view.tq ../../deps/v8/src/builtins/extras-utils.tq ../../deps/v8/src/builtins/frames.tq ../../deps/v8/src/builtins/growable-fixed-array.tq ../../deps/v8/src/builtins/internal-coverage.tq ../../deps/v8/src/builtins/iterator.tq ../../deps/v8/src/builtins/math.tq ../../deps/v8/src/builtins/object-fromentries.tq ../../deps/v8/src/builtins/object.tq ../../deps/v8/src/builtins/proxy-constructor.tq ../../deps/v8/src/builtins/proxy-delete-property.tq ../../deps/v8/src/builtins/proxy-get-property.tq ../../deps/v8/src/builtins/proxy-get-prototype-of.tq ../../deps/v8/src/builtins/proxy-has-property.tq ../../deps/v8/src/builtins/proxy-is-extensible.tq ../../deps/v8/src/builtins/proxy-prevent-extensions.tq ../../deps/v8/src/builtins/proxy-revocable.tq ../../deps/v8/src/builtins/proxy-revoke.tq ../../deps/v8/src/builtins/proxy-set-property.tq ../../deps/v8/src/builtins/proxy-set-prototype-of.tq ../../deps/v8/src/builtins/proxy.tq ../../deps/v8/src/builtins/reflect.tq ../../deps/v8/src/builtins/regexp-match.tq ../../deps/v8/src/builtins/regexp-replace.tq ../../deps/v8/src/builtins/regexp-source.tq ../../deps/v8/src/builtins/regexp-test.tq ../../deps/v8/src/builtins/regexp.tq ../../deps/v8/src/builtins/string.tq ../../deps/v8/src/builtins/string-endswith.tq ../../deps/v8/src/builtins/string-html.tq ../../deps/v8/src/builtins/string-iterator.tq ../../deps/v8/src/builtins/string-pad.tq ../../deps/v8/src/builtins/string-repeat.tq ../../deps/v8/src/builtins/string-slice.tq ../../deps/v8/src/builtins/string-startswith.tq ../../deps/v8/src/builtins/string-substring.tq ../../deps/v8/src/builtins/torque-internal.tq ../../deps/v8/src/builtins/typed-array-createtypedarray.tq ../../deps/v8/src/builtins/typed-array-every.tq ../../deps/v8/src/builtins/typed-array-filter.tq ../../deps/v8/src/builtins/typed-array-find.tq ../../deps/v8/src/builtins/typed-array-findindex.tq ../../deps/v8/src/builtins/typed-array-foreach.tq ../../deps/v8/src/builtins/typed-array-reduce.tq ../../deps/v8/src/builtins/typed-array-reduceright.tq ../../deps/v8/src/builtins/typed-array-slice.tq ../../deps/v8/src/builtins/typed-array-some.tq ../../deps/v8/src/builtins/typed-array-subarray.tq ../../deps/v8/src/builtins/typed-array.tq ../../deps/v8/third_party/v8/builtins/array-sort.tq ../../deps/v8/test/torque/test-torque.tq ../../deps/v8/src/objects/intl-objects.tq -o "/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated" -v8-root ../../deps/v8
1920
  ccache g++  -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj.host/icutools/deps/icu-small/source/i18n/number_longnames.o ../deps/icu-small/source/i18n/number_longnames.cpp '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DU_COMMON_IMPLEMENTATION=1' '-DU_I18N_IMPLEMENTATION=1' '-DU_IO_IMPLEMENTATION=1' '-DU_TOOLUTIL_IMPLEMENTATION=1' '-DU_ATTRIBUTE_DEPRECATED=' '-D_CRT_SECURE_NO_DEPRECATE=' '-DU_STATIC_IMPLEMENTATION=1' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/usr/include -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil  -pthread - [... cut ...]
1938
make[2]: Leaving directory '/workdir/openwrt'
1939
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
1940
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
1941
make[1]: Leaving directory '/workdir/openwrt'
1942
make: *** [world] Error 2
1943
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
1944
Error: Process completed with exit code 2.

blacknesswing, 2021-03-15:
tools/v8_gypfiles/run_torque.target.mk:13: warning: overriding recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/node-v12.21.0/out/Release/obj/gen/torque-output-root/torque-generated/../../deps/v8/src/builtins/proxy-tq-csa.cc'
好多这个

blacknesswing, 2021-03-16:
> 日志不够详细，要错误1上面那段的具体报错

   ERROR: package/feeds/packages/node failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1
 make[1] world
大佬是这个吗？

BabXhwZUNBeTV, 2021-03-16:
https://github.com/blacknesswing/R2s-openwrt 
github Actions 编译直接仓库地址给出来。
日志log 直接可以看 也可以上传文件 说个问题 为嘛藏着掖着呢。
[11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/6146174/11_Compile.the.firmware.txt)

或者把完整日志 贴到 paste.ubuntu.com https://paste.ubuntu.com/p/g6htK5vmJQ/

blacknesswing, 2021-03-16:
> https://github.com/blacknesswing/R2s-openwrt
> github Actions 编译直接仓库地址给出来。
> 日志log 直接可以看 也可以上传文件 说个问题 为嘛藏着掖着呢。
> [11_Compile the firmware.txt](https://github.com/immortalwrt/immortalwrt/files/6146174/11_Compile.the.firmware.txt)
> 
> 或者把完整日志 贴到 paste.ubuntu.com https://paste.ubuntu.com/p/g6htK5vmJQ/

不是藏着掖着 是真不懂 ，我去弄一下，谢谢大佬

BabXhwZUNBeTV, 2021-03-16:
> 不是藏着掖着 是真不懂 ，我去弄一下，谢谢大佬

日志已经帮你贴出来了。

blacknesswing, 2021-03-16:
感谢刚才那个大佬帮忙弄了日志

1715173329, 2021-03-17:
上游问题，暂时先做一个 workaround 处理。

closed 2021-03-17 by 1715173329