immortalwrt/immortalwrt #320: Can i compile ssr-plus without ipt2socks-alt ?

sajati, 2021-03-15:
i tried to compile for tp-link tl-mr3420 v5, with luci-app-ssr, only include shadowsocks + shadowsocksr + simple-obfs but it can't be done 
```$ make download
 make[1] tools/download
..
 make[2] -C package/ctcgfw/ipt2socks-alt download
    ERROR: package/ctcgfw/ipt2socks-alt failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

sajati, 2021-03-15:
also tried to re compile without ssr and then i have this error when execute `make download`, what is this mean and how to solve ?
```make[2]: Entering directory '/home/somebody/ImmortalWrt-21.02/package/lean/vlmcsd'
mkdir -p /home/somebody/ImmortalWrt-21.02/dl
SHELL= flock /home/somebody/ImmortalWrt-21.02/tmp/.vlmcsd-svn1113.tar.gz.flock -c '     echo "ERROR: No download method available"; false'
ERROR: No download method available
make[2]: *** [Makefile:52: /home/somebody/ImmortalWrt-21.02/dl/vlmcsd-svn1113.tar.gz] Error 1
make[2]: Leaving directory '/home/somebody/ImmortalWrt-21.02/package/lean/vlmcsd'
time: package/lean/vlmcsd/download#0.13#0.00#0.12
    ERROR: package/lean/vlmcsd failed to build.

AmadeusGhost, 2021-03-16:
Try with: `cd dl && rm rm vlmcsd-svn1113.tar.gz` and then re-run make.

1715173329, 2021-03-16:
@sajati Fixed in #320.

commit b059bd32 (2021-03-16): download.mk: fix typo error

Fixes: #320
closed 2021-03-16 by 1715173329