immortalwrt/immortalwrt #321: X86 5.10 出错

osxdiy, 2021-03-16:
CC       /media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/objtool/builtin-check.o
  CC       /media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/objtool/builtin-orc.o
  CC       /media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/objtool/elf.o
elf.c: In function 'read_symbols':
elf.c:407:8: error: implicit declaration of function 'gelf_getsymshndx'; did you mean 'gelf_getsym'? [-Werror=implicit-function-declaration]
  407 |   if (!gelf_getsymshndx(symtab->data, shndx_data, i, &sym->sym,
      |        ^~~~~~~~~~~~~~~~
      |        gelf_getsym
cc1: all warnings being treated as errors
make[8]: *** [/media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/build/Makefile.build:97: /media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/objtool/elf.o] Error 1
make[7]: *** [Makefile:59: /media/lu/OPENWRT/OpenioT/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.23/tools/objtool/objtool-in.o] Error 2
make[6]: *** [Makefile:68: objtool] Error 2
make[5]: *** [Makefile:1927: tools/objtool] Error 2

AmadeusGhost, 2021-03-16:
哪个分支？

osxdiy, 2021-03-16:
openwrt-18.06-k5.4

1715173329, 2021-03-17:
OpenWrt 上游问题，静候修复。

closed 2021-03-17 by osxdiy